wled/WLED #3181: File system corruption when adding JSON Preset

werkstrom, 2023-04-22:
### What happened?

Presets get corrupted. This is the second time in a few days it has happened so, better report it: I was adding JSON presets both times, i.e. JSON commands. The last time I was adding this exact JSON: `{"pixart":{"anim": ""}}`. I then get this screen on the presets page: 
[image]
Reboot does not fix the issue (predictably) 
When looking in the JSON it is clearly corrupt with strange chars some places. Fixing the JSON, (manually) fixes the issue (predictably).

### To Reproduce Bug

Adding JSON presets.

### Expected Behavior

For the file system not to corrupt the file 

### Install Method

Self-Compiled

### What version of WLED?

2303240

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Get this on the serial:

.pio/libdeps/esp32dev/LittleFS_esp32/src/lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
.pio/libdeps/esp32dev/LittleFS_esp32/src/lfs.c:1076:error: Corrupted dir pair at {0x0, 0x1}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-04-22:
Known issue. The corruption has been mitigated as much as WLED can do it but can still happen in special circumstances.
Duplicate of #2793

closed 2023-04-22 by blazoncek

werkstrom, 2023-04-22:
Allright. Not saying anything should be done then, but leaving this here *in case* anyone would like to address it in the future: Not sure how things are done in code today either so I could be out sailing, but: 

If we write to one single file we basically are out of luck should the write go wrong. We could use intermediate shadow files for editing and then swapping them in the end (at save) validating their integrity before committing. Something along then lines. On save:

1. Make a copy of (f.ex.)  preset.-json --> checkpreset.json
2. Add changes like today to preset.json (guessing arduinojson) 
3. Validate integrity of preset.json
- if OK delete checkpreset.json and continue on as usual. Job done should be 95% of cases.
- if NOT OK delete preset.json, copy checkpreset.json --> preset.json, start over at 2

Now, I realize I might be way out here so it's just a thought.

blazoncek, 2023-04-23:
The approach is good on paper but unfortunately it takes a lot of CPU cycles and requires enough free file system space.
presets.json may be as large as few 100s of kB. This may be an issue on constrained devices.

Unfortunately you cannot just make temporary presets.json and write new preset into it as presets are not kept in RAM.

werkstrom, 2023-04-23:
We could "move" the approach to the browser.

On save:
1. Get a fresh copy of current preset.json possibly update wleP (but not sure if that is complete or we need a new one).
2. Update preset.json exactly like today
3. If we are still in communication with the device (not sure what happens to the server on these occasions). Read the now updated presets.json and validate it. Quite possibly a simple "is this valid json" will do. If we are *not* in communication with the device we'd have to rely on the presetError function, see below.
4. If the validation fails, go back to step 2 and try again. Add some fail safe to only try a number of times and not perpetually. 
5. Done.

There is already an embryo to this in the "presetError" function but it kind of just tells the user "here's a string, do with it what you like" (from what I saw just glancing it). We could also extend that to offer to restore the file directly at that point.

This approach still means we don't let go of the task until we have ensured it has succeed, which IMHO is important. I could try this approach if you approve of it.

blazoncek, 2023-04-23:
1. presets.json may be big se explained above. And existing presets are stored in browser's JS variables already.
2. -
3. when preset is saved a new presets.json is loaded into browser. If corruption happens you have a backup copy in your browser's cache and you are prompted with this backup
4. in case of corrupt file it could be enough to upload RAM content back to file system as a restore operation (it is possible but not yet done)

werkstrom, 2023-05-03:
Some added context:
This issue appears quite regular. I only have a very limited amount of presets on this dev device and more or less every time I add one, the file gets corrupted. So this time it happened I did som documentation.

1. I had 4 presets allready
2. Trying to add a fifth one.
3. Simple, wanting to store a Palette, an effect and its settings.
4. clicking Add
5. Allowing everything to be "default"
6. Saving to preset 4 (first free as 1,2,3 and 5 is taken)
7. Setting the name to:"VER 1"
8. Click save
9. Gets presented with the "You don't have any presets" and the JSON.`

Some notes on that:

1. The *working* version is sorted in the key order
2. On the new preset: the *working* has a text "Please refresh..." where as the *non working* has a long series of "stop"-objects on the new preset.
3. The file is not corrupted, it's fully readable and the JSON validates
4. This issue happens a bit to often for me to be a fringe case
5. No serial output this time unfortunatly

SO
Here's: the JSON that is now in the preset.json:
```
{"0":{},"5":{"pixart":{"anim":"mario33"},"n":"Mario33","ql":"2"},"2":{"pixart":{"anim":""},"n":"Stop Animation"},"1":{"pixart":{"anim":"heart32"},"n":"Heart 32"},"3":{"on":true,"bri":128,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":144,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,160,0,0],[0,0,0,0],[0,0,0,0]],"fx":95,"sx":128,"ix":128,"pal":249,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"PopcornDefault","ql":"1"},"4":{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":35,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,160,0,0],[0,0,0,0],[0,0,0,0]],"fx":110,"sx":8,"ix":128,"pal":61,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"VER 1"}}
```

And here's the JSON that is presented in the UI:
````
{"0":{},"1":{"pixart":{"anim":"heart32"},"n":"Heart 32"},"2":{"pixart":{"anim":""},"n":"Stop Animation"},"3":{"on":true,"bri":128,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":144,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,160,0,0],[0,0,0,0],[0,0,0,0]],"fx":95,"sx":128,"ix":128,"pal":249,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":1},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}],"n":"PopcornDefault","ql":"1"},"4":{"n":"VER 1","win":"Please refresh the page to see this newly saved command."},"5":{"pixart":{"anim":"mario33"},"n":"Mario33","ql":"2"}}
```

blazoncek, 2023-05-03:
1. this is browser's cached version
2. actual preset is only modified on the device, not in browser
3. if you refresh (assuming you received "success" response to saving) the browser's cache should be updated
4. perhaps because your browser/device combo is not refreshing (I am guessing)
5. doesn't matter
6. if you verify/download presets.json via /edit, is it still corrupted and/or contains "Please refresh..."?

werkstrom, 2023-05-03:
6: It is _not_ corrupt, has _no_ "please refresh" but does _not_ work (according to the UI)

blazoncek, 2023-05-03:
After having "Please refresh..." go to UI settings and clear local storage (bottom of the page). Then return to UI and verify if you still see "Please refresh..."

If you do and the preset saved in the device contains "Please refresh..." then there must be some race condition in JS.

blazoncek, 2025-12-29:
Anyone stumbling on this: Presets on ESP32 get corrupted (99.999%) on save by incompatible flash mode used when flashing ESP for the first time.
To resolve the issue *you will need to flash new bootloader using DIO or DOUT mode*.