wled/WLED #3183: 404 on main page. skin.css referenced but not found

werkstrom, 2023-04-23:
### What happened?

Appears lacking graceful handling of non existent skin.css. No big issue, but still...
[image]

### To Reproduce Bug

Webpage + press f12

### Expected Behavior

Handle the missing file gracefully

### Install Method

Self-Compiled

### What version of WLED?

v0.14.0-b2 "Hoshi" (2303240)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-04-23:
skin.css is optional, uploaded by user.
You can disable loading of skin.css in UI preferences.

If you know how to detect if skin.css is present without making a request to web server, please add detection code to index.js

werkstrom, 2023-04-23:
Can't control the behavior of the browser as it has to do it's thing before letting javascript know. But modified slightly so the element is not added if it does not exist (thus getting rid of one error) and adding an explanation to the log.
[image]
proper handling of this would be to have the "server" provide info on if load should occur. The cfg.comp.css variable is true based on the config setting (i guess). So either it needs to return false if the file does not exist, but that might change other behavior else where, alternatively a new parameter needs to be created on the server that indicates the existence of the css or not. like cssExists. This might be good enough though since you now at least know why you get an error.

blazoncek, 2023-04-23:
I would prefer you do not mix different issues into one PR.

werkstrom, 2023-04-23:
Removed it.

blazoncek, 2023-05-27:
skin.css is optional, uploaded by user (or preloaded with board vendor) and as such may not exist but if it does, it has to be loaded.
404 is hidden from user unless developer tools are used.
As such resolution is unimportant.

closed 2023-05-27 by blazoncek