wled/WLED #3185: Deliberately Reboot Every X Seconds/Hours

danbates2, 2023-04-24:
One of the presets I'm using has a pattern which I like, but the pattern degrades over time until it's not very interesting at all. After two days it's pretty much at a stand still.

The pattern is called Phased if that's helpful for a bug report..

A timer function to reboot the ESP device every 24h for my setup would be ideal. I imagine it'd be useful in other context where a refresh of network stuff would be potentially helpful?

blazoncek, 2023-04-24:
Just use API for reboot in preset and addign such preset in "Time-controlled presets".
The trick is that when you try to save such preset WLED will reboot *before* save is finished.
To overcome that, you will need to edit *presets.json* using /edit page.

[Familiarize yourself](https://kno.wled.ge/interfaces/json-api/)) with custom JSON API.

Use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

closed 2023-04-24 by blazoncek

danbates2, 2023-04-25:
I've had a go and the JSON formatting is not human readable. That my limit for non-critical tasks :)

Thanks anyway

blazoncek, 2023-04-25:
use [JSON Formatter](https://jsonformatter.org/) to first prettify then minimize

danbates2, 2023-06-08:
I tried creating a reboot.json file in the SPIFFS editor but didn't have success. Could I be a complete pain and suggest an example and step-by-step be provided by the knowers? I can pay, workshop beautification is important :)

blazoncek, 2023-06-08:
You will need latest nightly build, then create a new preset with this content:
```json
{"rb":true}
```
When preset is saved get its ID and enter it into Time scheduled presets (Time & Macro Settings) to run at the desired time.