wled/WLED #3189: Feature request: Possibility to change preset on button press, not release

Sebbb, 2023-04-28:
**Is your feature request related to a problem? Please describe.**
I'm using WLED to control my bathroom light and to switch it on/off with a momentary switch. Right now, the light only toggles when I release the momentary switch (configured as "Pushbutton" in WLED), not when pressing it.

**Describe the solution you'd like**
I would like the LEDs to switch on/off when pressing the button down.

**Describe alternatives you've considered**
There is no good alternative except changing the source code (or accepting the current behavior) :)
I understand why the current implementation only triggers an action when releasing, and obviously this behavior should not break.
My idea is, when the same preset is configured for all actions (short, long, double), it makes no sense to wait for the release, and the action could trigger instantly.

**Additional context**
I already implemented it for me, but maybe it's useful for others as well.
See commit https://github.com/Sebbb/WLED/commit/d5e940a572b0edadea38f402d0ce7e3bd340bf5f
I didn't create a PR as I don't know if my idea how to solve it is good. My code probably isn't ;)

blazoncek, 2023-04-28:
Unfortunately in such case you do not know if double press is about to happen. Or if it is going to be long press.

If you've implemented it, please share your code.

Sebbb, 2023-04-28:
Hello,

that's the thing: If you configure the same preset for short/long/double press, it makes no difference and there is no need to wait. That's how I implemented it :)

Sebastian

blazoncek, 2023-04-28:
Make a PR and let's see if @Aircoookie approves.
IMO it is a viable compromise but will need to be documented in KB as well. Please prepare an update for WLED-docs too.

Sebbb, 2023-04-28:
As said, I don't think my code is very nice, so before putting time into documenting it, let's see if the overall idea is good (it's not very intuitive, a dedicated setting would be easier to understand), and if somebody can implement it in a less hackish way

blazoncek, 2023-04-29:
As you wish. I, personally, would love to see more contributions like this.
IMO it is not *hackish* that much if it is documented well.

If we'd add a checkmark/option to trigger on the "rise edge" as opposed to the "fall edge" then we would need to inform users that in such case long press and double press are no longer working. A worse situation IMO.

Sebbb, 2023-06-02:
Implemented with https://github.com/Aircoookie/WLED/pull/3226 - thanks :)

closed 2023-06-02 by Sebbb