wled/WLED #3191: Sunset at 00:00 - lights not turning on at sunset

Illumidel, 2023-05-01:
### What happened?

[image]

### To Reproduce Bug

Time zone : US-EST/EDT
Offset : 0
Latitude : N    45,60
Longitude : W   73,44

### Expected Behavior

Sunset will set to : 00:00

### Install Method

Self-Compiled

### What version of WLED?

13.3 and 14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

If I switch Latitude to 46 instead of 45,60 everything goes back to normal . 
This started 2 days ago

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-05-01:
Known issue due to custom sin/cos functions developed for speed and memory footprint.
It will go away after a day or two or if you enter slightly different Lat & Lon.
Closing.

closed 2023-05-01 by blazoncek

blazoncek, 2023-05-01:
WORKAROUND: if you compile with PIO and add *build_flag*`-D WLED_USE_REAL_MATH` then WLED will use standard *libc* functions instead