wled/WLED #3196: Always Turns On Orange

ElToberino, 2023-05-05:
### What happened?

At ESP32 startup /reset all leds light up in bright orange for a short moment before the "Apply Preset xx at boot" and the brigthness limiter take place. This is a problem for my setup as I enabled the brightness limiter due to a limited power supply. The initial orange draws to much current and causes a "reset loop" of the psu.
This problem was also described in the [discourse forum](https://wled.discourse.group/t/always-turns-on-orange/3307).

### To Reproduce Bug

Start up ESP32 with an "Apply Preset xx at boot" setting other than orange.

### Expected Behavior

No orange flash at startup - first light up of the leds with the "Apply Preset xx at boot" setting.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b2 (build 2304280)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Environment: 
ESP32 board (similar DEVKIT V1).
1536 leds attached (768 on GPIO 16, 768 on GPIO 4).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-05-05:
I have the fix ready, but is not yet published.
If you are willing to test it, please join Discord and tag me in beta testing channel.

Posiek, 2023-05-06:
I have similar problem recently. It look like WLED reset and after startup it light up in bright orange. Yesterday LEDs was turn of and at 4:20 AM turned on its own. Today it turn off randomely and then after 20 minutes turn on again and light up with orange. I thought that something is connected in not proper way?

blazoncek, 2023-05-06:
Please use Discord where I posted binaries with the fix included.

moose4lord, 2023-06-20:
Sorry, but it's not clear to me if this has been fixed in the latest code. I downloaded and compiled the current main branch from GitHub, but I still see the bright orange flash when the strip is first turned on. It also flashes when I save anything on the config menus. #3220 seems to imply the fix has been merged with main, but I don't see it.

blazoncek, 2023-06-20:
@moose4lord you may have other issues then.

moose4lord, 2023-06-20:
It seems very similar to the issue @ElToberino raised. I have an ESP32 with a  strip of 64 WS2812s. In LED Preferences I have the "Turn LEDs on after power up/reset" checkbox checked, but the default brightness set to 16, so I don't scorch my eyeballs when it turns on. If I reset the ESP32, it briefly flashes orange (as if brightness=128), then resets to the correct intensity (brightness=16).

The flash is occurring the very first time ```strip.service();``` is called at line 103 of wled.cpp.

blazoncek, 2023-06-20:
@moose4lord that is the expected behaviour. /cc @ElToberino 

The boot sequence is as follows:
- if "Turn LEDs on at boot" is checked, light up LEDs with orange color, otherwise with black
- if "Apply boot preset" is anything but 0, apply that preset when it is due, oherwise skip applying preset.

Presets are applied in asynchronous fashion since 0.14.
If you do not want brief orang flash, uncheck "Turn on LEDs at boot" and only apply boot preset.
Presets *can* have On/Off state saved in them. There is no need to have "Turn on LEDs at boot" checked if you have boot preset assigned.

I am closing this issue as it is a configuration issue, the core bug for unwanted orange flash with "Turn on LEDs at boot" unchecked was fixed in #3220

closed 2023-06-20 by blazoncek

Andrei-Iosifescu123, 2023-10-15:
> @moose4lord that is the expected behaviour. /cc @ElToberino
>
> The boot sequence is as follows:
>
> * if "Turn LEDs on at boot" is checked, light up LEDs with orange color, otherwise with black
> * if [...]

Even after disabling 'Turn on LEDs at boot' the leds still flash for half a seconds before applying the preset.

blazoncek, 2023-10-15:
Half a second? I'd like to see that!
@Andrei-Iosifescu123 please post your configuration and please use Discord or WLED [forum](https://wled.discourse.group) for further communication.

Andrei-Iosifescu123, 2023-10-15:
> Half a second? I'd like to see that! @Andrei-Iosifescu123 please post your configuration and please use Discord or WLED [forum](https://wled.discourse.group) for further communication.

As you can see [here](https://imgur.com/a/ANvGemH), the strip quickly flashes and then applies the default preset. I would like to get rid of the flashing.

I'm using an ESP8266 D1 Mini with 3 bd441 transistors driving an analog rgb strip connected to pins D6 (R), D5 (G) and D7 (B).

blazoncek, 2023-10-15:
That's not half a second. 😀  It may be a hardware limitation of PWM drivers as some GPIO pins may use internal pull-up resistors at boot.

Andrei-Iosifescu123, 2023-10-15:
> That's not half a second. 😀 It may be a hardware limitation of PWM drivers as some GPIO pins may use internal pull-up resistors at boot.

Yeah, but still kinda annoying. I tried connecting a 470uF cap between the base of the transistor and GND and it worked, but interfered when using LEDfx for syncing to music. Maybe the value is too big, i'll try some smaller ones.

mechalv, 2023-11-05:
Hi, any news on this? I’m having this exact same issue with a Dig-Uno (on 0.14.0) on my setup. I’ve got a relay that triggers the ws2811 LEDs on parallel to the WLED and there is a very bright orange flash of less that 0,5s before launching my playlist. My goal is to have a measured sweep effect before a solid preset, so having firstly a bright flash/blink has stopped my project.
I’ve seen references to similar problems in #346 and #3430 

In my case I’m also experiencing a stutter on the sweep effect of that playlist, so I’m having to delay it with a 3s solid black preset. I’m not very sure if this is only a problem of mine and if it’s worth opening an issue.
Thanks.

blazoncek, 2023-11-05:
Use transition time of 0 and do not set LEDs to turn on at boot.

mechalv, 2023-11-07:
> Use transition time of 0 and do not set LEDs to turn on at boot.

Woah!! That did the trick, thank you very much, no more orange flash at boot and the playlist still works!! It’s interesting, so maybe WLED should set that setting off automatically if a preset different to 0 is set for the boot sequence?
The stutter at boot is still there so I might open an issue.

Andrei-Iosifescu123, 2023-11-09:
> Woah!! That did the trick, thank you very much, no more orange flash at boot and the playlist still works!! It’s interesting, so maybe WLED should set that setting off automatically if a preset [...]

I fixed the problem by adding 10K resistors between the source and the gate of the MOSFETs.

blazoncek, 2023-11-09:
> I fixed the problem by adding 10K resistors between the source and the gate of the MOSFETs.

That is a mandatory thing.

Andrei-Iosifescu123, 2023-11-09:
> > I fixed the problem by adding 10K resistors between the source and the gate of the MOSFETs.
> 
> That is a mandatory thing.

Figured out the hard way 😁

Andro-Marian, 2024-11-01:
Have this may break the other thing?

I flashed the 0.15.0-b6 and when turning off/on the bulbs (7w or 15w) has a weird flicker when using the power putton from the web interface. Not sure why. When the fade-in starts and when fade-out is finishing.
https://drive.google.com/file/d/1YgvZzD1sl7MFx1y-zQKS_JEvJhUdDfPS/view?usp=sharing

Andro-Marian, 2024-12-14:
> Have this may break the other thing?
>
> I flashed the 0.15.0-b6 and when turning off/on the bulbs (7w or 15w) has a weird flicker when using the power putton from the web interface. Not sure why. When [...]

This still not fixed in the release.