wled/WLED #3197: Game of Life 2D

SimonePDA, 2023-05-07:
### What happened?

The engine that calculates the neighbours is not respecting the rules:
One pattern generates a horizontal glider where it should turn into a traffic light

[image]

the second step from right to left is missing one cell because the right one is this

[image]

moreover, when the generation ends with oscillators and gliders it never restarts and it becomes stuck in an infinite not interesting evolution, The CRC calculation does not work well in these cases.

### To Reproduce Bug

It happens as a result of the creation of any of the patterns in the sequence during the normal generation process. If any of the patterns of the sequence is the last in the game, then it loops forever. due to

### Expected Behavior

Evolve in the right Game of Life pattern

[image]

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1 "Hoshi"

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

You can sumulate the correct behaviour here

https://playgameoflife.com/

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-05-13:
There is a bit of randomness in reproduction decisions to avoid "gliders" (together with CRC).
While I was testing it, I did not have any endless gliders.

blazoncek, 2023-05-27:
@SimonePDA have you tested GoL as implemented in WLED 0.14.0-b2?
And when a *glider* is seen the cycle should restart within 10s.

SimonePDA, 2023-05-28:
I have switched to the B2 (14.0.b2) build and indeed the GoL restarts always after a while, even if gliders are the only thing left.

I would say that this version fixes this specific issue. Can't say anything about the wrong generation that caused a non existing glider because it never happened again and I have no way of forcing a pattern on the display and let it evolve.

Thanks for your attention.
Simone

closed 2023-05-28 by SimonePDA