wled/WLED #3198: ESP32 crashes sometimes when changing preset

ElToberino, 2023-05-07:
### What happened?

In my (relatively) complex 2D-setup I often get resets/crashes of the ESP32 when changing the preset via webpage. 
It is interesting, that the crashes don't happen, when I run the (same) effects in a playlist inderintely repeating - so I assume it's somehow related to webserver or Wifi as without interaction with webserver the crashes don't happen.

In debug mode I get the following error message:
```
Sending WS data Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x401ad67d  PS      : 0x00060a30  A0      : 0x800e64b9  A1      : 0x3ffb1e60
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x8008b6a0  A5      : 0x3ffdc930
A6      : 0x00000000  A7      : 0x3ffb0060  A8      : 0xffffffff  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00000005  A13     : 0x00000020
A14     : 0x00000020  A15     : 0x00000621  SAR     : 0x0000000a  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x40002390  LEND    : 0x4000239f  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x401ad67a:0x3ffb1e60 0x400e64b6:0x3ffb1e80 0x400de8d5:0x3ffb1eb0 0x400dfcf5:0x3ffb1ee0 0x400ea013:0x3ffb1f30 0x40112969:0x3ffb1f50 0x40112c7e:0x3ffb1f90 0x4013148d:0x3ffb1fb0 0x4008b4e6:0x3ffb1fd0

Rebooting...
```

### To Reproduce Bug

If necessary I can provide my cfg and preset files.

### Expected Behavior

No crashes.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b2 (build 2304280)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

Environment:
ESP32 board (similar DEVKIT V1).
1536 leds attached (768 on GPIO 16, 768 on GPIO 4).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-09-05:
Hi, we have added some stability improvements in version 0.14.0-b4 (beta 4). Please check if you still see crashes when changing presets, when using the latest beta.

blazoncek, 2023-09-06:
And also add exception decoder and try to capture function call trace.

softhack007, 2023-10-02:
Closed due to missing feedback from original problem author.

closed 2023-10-02 by softhack007

vdrummer85, 2023-11-10:
I'm seeing similar behavior using 0.14.0 and an ESP32-WROOM.  Were the stability improvements referenced in 0.14.0-b4 incorporated in the final 0.14.0?

baffles, 2023-12-05:
@softhack007 / @blazoncek: I agree with @vdrummer85, I see the same thing on 0.14.0. It's pretty simple to reproduce by adding some presets with different segment configurations. If you keep switching between them, you'll get a crash in fairly quick order.

blazoncek, 2023-12-05:
Please compile with debug option and monitor serial output. Post crash report using exception decoder.

Add `monitor_filters = esp32_exception_decoder` to your build environment and use `-D WLED_DEBUG` build flag.

Royoga, 2024-01-04:
Hello,
we experiencing the same issue.
We added `monitor_filters = esp32_exception_decoder` and `-D WLED_DEBUG `
and got he following log

`
SON buffer locked. (11)
JSON buffer released. (11)
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401b9952  PS      : 0x00060430  A0      : 0x800eb1be  A1      : 0x3ffb1ed0
A2      : 0x3ffdb82c  A3      : 0x3ffb1f04  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x0000012d  A7      : 0x3ffbe0d9  A8      : 0x0000e302  A9      : 0x0000d108
A10     : 0x3ffdfef4  A11     : 0x00000001  A12     : 0x00000258  A13     : 0x00000258  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000e318  LBEG    : 0x401b97ce  LEND    : 0x401b97e3  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401b994f:0x3ffb1ed0 0x400eb1bb:0x3ffb1ef0 0x40118b58:0x3ffb1f50 0x401190b6:0x3ffb1f90 0x40137fdd:0x3ffb1fb0 0x4008bfa6:0x3ffb1fd0
  #0  0x401b994f:0x3ffb1ed0 in Segment::restoreSegenv(Segment::TemporarySegmentData&) at wled00/FX_fcn.cpp:418
  #1  0x400eb1bb:0x3ffb1ef0 in WS2812FX::service() at wled00/src/dependencies/json/ArduinoJson-v6.h:4811
  #2  0x40118b58:0x3ffb1f50 in WLED::loop() at wled00/wled.cpp:115
  #3  0x401190b6:0x3ffb1f90 in loop() at ***/wled00/wled00.ino:20
  #4  0x40137fdd:0x3ffb1fb0 in loopTask(void*) at ***\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #5  0x4008bfa6:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c

---WLED 0.14.0 2310130 INIT---
esp32 v3.3.6-16-gcc5440f6a2
arduino-esp32 v1.0.x

CPU:   ESP32-D0WDQ5 rev.3, 2 core(s), 240MHz.
FLASH: 4MB, Mode 3 (DOUT), speed 40MHz.
heap 281724
PIN ALLOC: Pin 1 successfully allocated by 0x89 (137)
PIN ALLOC: Pin 2 successfully allocated by 0x8a (138)
Registering usermods ...
heap 281556`

We are on V0.14.0.
Any ideas how we could proceed?

blazoncek, 2024-01-04:
Resoved in 0.14.1-b2