wled/WLED #3201: Allow WLED to be used through a Proxy

weazel91, 2023-05-12:
**Is your feature request related to a problem? Please describe.**
I'm running a version of FPP (https://github.com/FalconChristmas/fpp) to which has an inbuild proxy. This allows us to connect to LED Controllers though this device allowing remote access to WLED. This is possible wth FPP to the home page, however when trying to access anything other than the homepage, for example the settings page, it tries to revert to the IP address of the FPP device.

**Describe the solution you'd like**
WLED should, when visiting pages visit the URL it's already on rather than the IP address of the FPP device. At present if I am on the following URL:

http://10.10.10.1/proxy/192.168.1.1

with 10.10.10.1 being the IP of FPP and 192.168.1.1 being the IP of WLED if I click settings in WLED the URL gets rewritten to:

http://100.98.153.9/settings

this should instead be:

http://10.10.10.1/proxy/192.168.1.1/settings

**Describe alternatives you've considered**
If you manually put the address above (http://10.10.10.1/proxy/192.168.1.1/settings) you can access the page as normal, ideally navigating though a proxy should be possible.

**Additional context**
N/a

Thank you for your ideas for making WLED better!

softhack007, 2023-05-13:
Seems to be a duplicate, or at least very similar to these requests:
* https://github.com/Aircoookie/WLED/issues/3069
* https://github.com/Aircoookie/WLED/pull/2788

blazoncek, 2023-05-15:
Closed as a duplicate of #3069

closed 2023-05-15 by blazoncek