wled/WLED #3202: Add more segments then just 10

icepick3000, 2023-05-13:
The segments feature is very powerful.. I would only like to be able to make more segments since my setup will contain thousands of leds.

blazoncek, 2023-05-13:
Use ESP32 and you can set-up up to 32 segments.

icepick3000, 2023-05-14:
Let me try to explain what I am trying to do.. maybe there is another way.. 
Lets say I have a matrix of 100x1024 leds and I am displaying an effect on it (for example the builld in chun chun effect). 
Then I want to display a symbol (lets say a circle) but I want the effect to still run for the leds not incuded in the symbol. The way I do it now is make a square segment in the middle (the chun chun effect still runs on the other segements just fine) and in the middle I will display the circle symbol. Around the circle the chun chun is not visible (its a different segment now). What I want is to have the whole display show chun chun but except for those few leds that are part of the circle. 

I am basically looking for a partially transparant segment. 
So I was thinking to map every led of the symbol as a separate segment but then even with 32 it won't be enough. But maybe there are better way to do what I want to do.

blazoncek, 2023-05-14:
First off, WLED, even on ESP32, can only handle about 8000 LEDs as this is hardware limitation of the microcontoller used (RAM, CPU power). So 100x1024 is not possible.

Second, what you are trying to achive is possible but only with certain effects that allow overlaying segments. Not every effect allows this.

Lastly, each segment consumes RAM. Plenty of it. Increasing available segments will deplete it sooner. On top of that each segment can run different effect causing more CPU demands.

You may be better off by using external software like xLights, LedFx, Jinx or similar, hosted on a proper PC, for the effect you are after.

Please use WLED [forum](http://wled.discourse.group) or Discord for help and support questions.

closed 2023-05-14 by blazoncek