wled/WLED #3203: WLED 0.14.0-b1/2212222 Will NOT connect to WiFi

chilled35, 2023-05-16:
### What happened?

Flashed my ESP8266 I've been using very successfully with 0.13, wanting to use the 2D matrix.  Upon successful flash, trying to get the ESP to connect to my wifi is impossible, it sits there 30s and eventually times out.

### To Reproduce Bug

Flash 0.14, try to attach to WiFi, fails, 100% of the time. See attached screen grab of failure.

### Expected Behavior

Connect to the Wifi as it did with 0.13

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b1/2212222

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
None available.
```

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-05-18:
what is the result if you flashed the latest  ? if you cant compile from source then use  https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware

dosipod, 2023-05-18:
I think i have some of those old ESP8266 that do not work with 14 , this is one of hem 
[image]
The same module has also being reported by others as it does not work .

chilled35, 2023-05-18:
So I tried a couple of variations on the theme of trying to flash these devices ( https://www.aliexpress.us/item/3256805229189643.html ) with 0.14 b1 without and with ethernet option, no cigar.  Tried 0.13.3 and 0.13.1 and they both with just dandy.

Bizarrely, all ten D1 Mini boards arrived last week and my first job was to flash them all with ESPHome to detect the MAC addresses of each so I could handwrite the last 4 chars of the address to make identifying each board by eye easier.  All ten boards flashed fine, connected to the wifi fine, appeared as adoptable boards in Home Assistant/ESPHome, so I have no real reservations in saying all the boards are completely functional.  Just seems 0.14b1 for whatever reason, is a bit temperamental with Wifi, oddly.

chilled35, 2023-05-18:
I haven't had the time to download/compile/install the code from github manually... will try over the weekend...

blazoncek, 2023-05-27:
Closing as WiFi is working correctly on good quality ESP8266 devices.

closed 2023-05-27 by blazoncek

chilled35, 2023-06-27:
Quality isn't the issue.  It boils down to the components used on differing versions of the ESP8266 D1 Mini. If you have the version with the metal can module, it works fine.  If you have the PCB built with discrete components, for whatever reason, it does not work. What the difference in circuit between the two is, I have no idea.