wled/WLED #3204: JSON APIs assigned to buttons  , e.g  {"ps":"1~4~"}

dosipod, 2023-05-17:
### What happened?

We normally used {"ps":"1~4~"}  in a preset and that will work in ui but  not when assigned to buttons it does not work    only   win&PL=~&P1=1&P2=4 works from both ui and when assigned to buttons.

I am not sure that is limited to the above api as assigning other api to buttons like  {"on":"t"} will  work . I will try to look for more if that is only limited to this 

### To Reproduce Bug

Add  {"ps":"1~4~"}  to preset and assign to button action  ( clicking on preset will work but button action will not work ) 
Add  win&PL=~&P1=1&P2=4  to preset and assign to button action  ( clicking on preset will from ui and button action  ) 

### Expected Behavior

 {"ps":"1~4~"}    and win&PL=~&P1=1&P2=4  is expected to behave the same 

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b2 (build 2305090)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
More  more info on the same  are here  https://wled.discourse.group/t/physical-button/8289
and here https://discord.com/channels/473448917040758787/473448917543944193/1108240349115191386
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-05-21:
Please check if "Send notifications on button press or IR:" is enabled in Sync settings.
Unless this setting is checked pressing button will not update UI and WLED state.

This is by design.

dosipod, 2023-05-22:
If "Send notifications on button press or IR:"   is enabled in Sync settings then other lights on the network will react and still pressing button will not  work correctly for  json  but will work fine with win&PL=~&P1=1&P2=3 , I have attached a pic of the four tests I did with and without changes to Sync settings with both  format . I only use the MCU on board GPIO0 button if that could make a difference .  If you see otherwise then could be something from our setup 
[image]

blazoncek, 2023-05-22:
Then we may need to wait for @Aircoookie to look into it as notification logic is still a bit mystery to me.

blazoncek, 2023-05-22:
I have traced the workings and it looks like the JSON preset cycling will not work with buttons due to the way how button calls preset.

From your example:
- preset 4 contains "ps":"1~3~" and is assigned to a button
- when button is pressed it *schedules* applying preset 4 (presets are asynchronous since 0.14)
- when the (async) schedule is due it loads preset 4 from FS and removes "ps" property to prevent recursive crash
- resulting preset is empty so execution does nothing

blazoncek, 2023-05-31:
I have an idea!
Will provide binary for testing soon (via Discord).

PR #3238 (2023-06-13): Beta 3
Adds the following:
-   Bumped version to 0.14-b3 (beta 3)
-   Reverse proxy support (replaces #2788 and fixes #3069 and #1177)
-   PCF8754 support for Rotary encoder (requires wiring INT pin to ESP GPIO)
-   Rely on global I2C pins for all usermods (BREAKING CHANGE; removed individual I2C initialisations and pin set up)
-   adds better UX selecting GPIO pins in settings
-   various fixes and enhancements (fixes #3204 #3236)
closed 2023-06-13 by blazoncek