wled/WLED #3205: Fix MQTT Payload size limition

dosipod, 2023-05-17:
We normally use MQTT to avoid despondency on Home assistant , having generic solution like MQTT will help  provide more options and it is a door to more advanced control  . 

For the case in point using a small matrix of 16x16 we are seeing limitation with pixel art  sent via MQTT as only part of the  image is displayed . We were advised by @blazoncek  to try to tweak the -DMQTT_MAX_PACKET_SIZE=1024  flag  in platformio.ini  but nothing we tested worked and we see no difference so might be we are missing another flag or we have no way to verify if there is limitation from the source ( I did try multiple mqtt clients and node red  with the same result ) .

It is worth mentioning that @werkstrom  pixel art converter still works if we send the api directly do wled device or copied the same api to a preset . 

I have also tried to seek other devs hits to see if we are missing something but it seem that is a bit of uncharted area and might impact stability as mqtt with wled could  be impacting other areas as it was mentioned by other forks like SR  but I hope that is no longer the case    

I will be attaching only one  example but if that is going to happen and more info is needed then we can provide more examples and details , else if the subject matter is low priority then we could come back to that later 

[image]

blazoncek, 2023-05-17:
Looks like WLED will need to implement partial packet handling, which it doesn't ATM.
https://github.com/marvinroger/async-mqtt-client/issues/98

blazoncek, 2023-05-23:
Fixed by 56a3bff42edaa0eaa6ccdfe9a1e789c1abd3f262

closed 2023-05-23 by blazoncek