wled/WLED #3206: UI Bug iOS App (effects tab)

gregorne, 2023-05-17:
### What happened?

When trying to select effects though the mobile (iOS) app, the effects list seems to be misaligned. Also tapping on the effects does not select the effect where the screen is touched but a different one.
See image and video below

[image]

https://github.com/Aircoookie/WLED/assets/27886137/78e3659e-6130-4842-a411-1e50eab95e5e

### To Reproduce Bug

Open the wled app, select a device, go to effects tab

### Expected Behavior

„Normal“ nicely aligned effects list with correct touch alignment. 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b1 (build 2212222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-05-18:
You can use wled native  iOS  app https://apps.apple.com/us/app/wled-native/id6446207239

gregorne, 2023-05-18:
Thank you! I’ll check it out. 
would be nice if the „official“ app fixes the bug

closed 2023-05-18 by blazoncek

marcopinder, 2023-06-01:
This shouldn't be closed as the 'Developer Website' link in the Apple App Store listing for the WLED app is a link to this very repository. Where else should this issue be logged?

softhack007, 2023-06-02:
> Where else should this issue be logged?

Here: https://github.com/Aircoookie/WLED-App/issues