wled/WLED #3207: Feature request:  DDP RGB+CCT (network)

Sebbb, 2023-05-18:
Hello,

my bathroom has several different RGB+CCT stripes, which I can't all connect to just one ESP32. So it would be nice to be able to use multiple ESPs for that. Unfortunately, the remote ones only display cold white when being controlled remotely (when using DDP RGBW).

So it would be nice to also have a DDP RGB+CCT option, which behaves similar to the already existing ones, but also transmits the white channel information.

blazoncek, 2023-05-19:
CCT handling is not in DDP specification. So, unless you talk to DDP authors and have them include CCT handling in DDP protocol it cannot be implemented.
At least no t correctly and future-proof.

Sebbb, 2023-05-19:
Hey @blazoncek, thanks for pointing that out.. I now sent a mail to the guy behind DDP and asked if it's possible to put that into the protocol definition. I'll close that feature request for now :)

closed 2023-05-19 by Sebbb

blazoncek, 2023-05-19:
I am not sure the CCT is indeed suited to be transmitted over DDP (or any other network protocol) as it is effectively white with different *tint*. Because you can transmit *tint* in two ways: a) two white channels (IMO bad) or b) white channel and (color temperature). Normally color temperature doesn't change (at least I see no reason for it to change in a per-frame fashion) during streaming (which DDP is) so there is no need to include it in the stream.

WLED internally handles CCT as a white channel and color temperature of that white (on a per segment basis).

Sebbb, 2023-05-19:
Does it maybe make sense then to add another (proprietary) protocol beside DDP to only sync/send the color temperature?
And then to select something like "DDP-RGBW + additional color temperature sync (network)"?