wled/WLED #3209: RGB+CCT strip: LEDs go dark when using Alexa to change light to "white" when "Auto-calculate white channel from RGB" is set

Sebbb, 2023-05-21:
### What happened?

I'm using RGB+CCT strips and have set "Auto-calculate white channel from RGB" to "Accurate".

When I tell Alexa to change the light color to white, it just goes off. I figured it sets col to [0, 0, 0, 255], but it seems the last value isn't used in that mode, so the LEDs just go dark.

### To Reproduce Bug

Use a PWM RGB+CCT LED strip, set "Auto-calculate white channel from RGB" to "Accurate" and tell Alexa to switch the light color to white.

### Expected Behavior

When I ask for white light while the LED strip is on, it should turn white and not dark :)

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b2 (build 2305090)

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Relevant log/trace output

```shell
Alexa sends:

PUT /api/2BLEDHardQrI3WHYTHoMcXHgEspsM8ZZRpSKtBGr/lights/1326925184/state HTTP/1.1
Host: 192.168.6.213
Accept: */*
Content-Type: application/json
Content-Length: 20

{"on":true,"ct":284}

---

State update:

{"state":{"on":true,"bri":255,"transition":0,"ps":-1,"pl":-1,"nl":{"on":false,"dur":60,"mode":1,"tbri":0,"rem":-1},"udpn":{"send":false,"recv":true},"lor":0,"mainseg" :0,"seg":[{"id":0,"start":0,"stop":1,"len":1,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":50,"col":[[0,0,0,255],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix": 128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0}]},"info":{"ver":"0.14.0-b1","vid":2212222,"leds": {"count":1,"pwr":0,"fps":3,"maxpwr":0,"maxseg":16,"seglc":[7],"lc":7,"rgbw":false,"wv":2,"cct":4},"str":false,"name":"WLED","udpport":21324,"live":false,"liveseg":-1,"lm" :"","lip":"","ws":3,"fxcount":187,"palcount":71,"cpalcount":0,"maps":[0],"wifi":{"bssid":"14:CC:20:AC:5F:92","rssi":-58,"signal":84,"channel":11},"fs":{"u":40,"t":499,"pmt":1684638572},"ndc":2,"arch":"esp8266","core":"2_7_4_7","getflash":2097152,"buildtime":"2023-04-24 14:49:54","lwip":1,"freeheap":15048,"uptime":496,"opt":111,"brand":"WLED","product":"FOSS","mac":"80646F9E2E8B","ip":"192.168.6.213"}}
```

### Anything else?

It seems that just calling `strip.setCCT(k);` isn't enough to set the color, it also needs to check the "Auto-calculate white channel from RGB" state.

I fixed it for myself:

```
diff --git a/wled00/alexa.cpp b/wled00/alexa.cpp
index c122402a..9b2db811 100644
--- a/wled00/alexa.cpp
+++ b/wled00/alexa.cpp
@@ -105,8 +105,15 @@ void onAlexaChange(EspalexaDevice* dev)
                        uint16_t k = 1000000 / ct; //mireds to kelvin

                        if (strip.hasCCTBus()) {
+                               uint8_t aWM = Bus::getGlobalAWMode();
+
                                strip.setCCT(k);
-                               rgbw[0]= 0; rgbw[1]= 0; rgbw[2]= 0; rgbw[3]= 255;
+                               if (aWM != RGBW_MODE_DUAL && aWM != RGBW_MODE_MANUAL_ONLY && aWM != AW_GLOBAL_DISABLED) {
+                                 rgbw[0]= 255; rgbw[1]= 255; rgbw[2]= 255; rgbw[3]= 0;
+                                 dev->setValue(255);
+                               } else {
+                                 rgbw[0]= 0; rgbw[1]= 0; rgbw[2]= 0; rgbw[3]= 255;
+                               }
                        } else if (strip.hasWhiteChannel()) {
         switch (ct) { //these values empirically look good on RGBW
           case 199: rgbw[0]=255; rgbw[1]=255; rgbw[2]=255; rgbw[3]=255; break;
```

But I guess that's not the whole solution, as it only checks the global override state, not the state of each LED.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Sebbb, 2023-05-21:
It's also somehow related to https://github.com/Aircoookie/WLED/issues/780, although a different use case (I have LEDs with white channels, but still calculate the white channel from RGB, so `if (strip.hasCCTBus())` doesn't help.

softhack007, 2023-05-21:
@Sebbb I've just fixed the math accidents in espalexa that lead to **NaN**, please check if it works (a bit) better now.

Your other fixes in alexa.cpp still make sense, just I don't have enough knowledge of the related code to decide if that's the "proper way" to solve the issue. Please make a PR so that @Aircoookie or @blazoncek can take a look.

Sebbb, 2023-05-21:
@softhack007 Hey, thanks for your quick help. I just compiled and tried it, together with my change it works well.
Although I'm not a fan of creating PRs for code I'm not happy with, I'll create one in the hope somebody will improve my code. I'm not a good C-coder :D
I'll close #3208 as it's solved :)

Sebbb, 2023-05-31:
Resolved by https://github.com/Aircoookie/WLED/pull/3211

closed 2023-05-31 by Sebbb