wled/WLED #3212: Extend the backup functionality to inculde files under wled-ip/edit

dosipod, 2023-05-22:
I think the ability to take a backup is important  and good enough as it is . But at the moment new features has being added to wled that is not covered by the current  backup functionality . Ledmaps , json animation ..etc will all have to be manually backed up .
It would be nice if we can  dump wled/edit content as one file for backup and then restore the same way . Not asking for an api to also do that but that would also be nice .

Just to clarify I am  doing that purely to enhance wled and we do I  not have any real life use case to backup that frequently  ( well we do a lot of destructive tests  and it will make our life a bit easier )  . If that is will not be considered then we could script it from our side so not the end of the world as  the firmware backup  script is a good example

dosipod, 2023-05-22:
Yeah that is the way I backup now , I have a folder with each MCUs content . I was thinking to modify this script https://github.com/Aircoookie/WLED/blob/main/tools/multi-update.sh    so instead of an update it will do download , so no worries you may close or keep as a reminder

dosipod, 2023-07-01:
You might have the scripts already :)  https://discord.com/channels/473448917040758787/473450515385024533/1005724880043507774

blazoncek, 2023-07-02:
Indeed I have, but those are mine only as they are a bit involved.
Not yet ready for primetime.

Basicaly they are extensions of the update scripts.