wled/WLED #3214: V0.14.0-b2 2D colour pallet and wheel not available

ceboxsell, 2023-05-23:
### What happened?

When setting up an 8 * 32 panel in 2D settings WLED does a soft restart of some type I assume as the LEDs reset to default.
If I select any effect e.g. Android I have the full colour wheel and pallet. If I then reset from Info or reset power when it boots the colour wheel and pallet are gone for good.
Change back to 1D and restart the colour options return.
Note version downloaded from https://install.quinled.info/

### To Reproduce Bug

Enable 2d and restart WLED with no previous preset enabled. If there is a preset before the 2d change when you select it the colour wheel etc return and remain until you restart WLED.

### Expected Behavior

Normal access to colour wheel and pallet where applicable

### Install Method

Self-Compiled

### What version of WLED?

V0.14.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-05-27:
Cannot reproduce.
Please erase flash and upload the latest 0.14 (or newer) available preferably using esptool or official web installer.
Then clear browser's cache and local storage.
Closing as there is no bug within WLED regarding disappearing color wheel or palettes.

closed 2023-05-27 by blazoncek

ceboxsell, 2023-05-28:
I did all of that. I tried on multiple devices same issue so not cache.
See if this still happens in later releases then. thanks.

blazoncek, 2023-05-28:
The only unaccounted for reason for missing wheel and palettes is incomplete load of page. Open browser's developer tools and monitor page load. The only 404 should be skin.css.

If you only use white LEDs then those will hide wheel and palettes as white LEDs cannot use color.

ceboxsell, 2023-05-28:
Yes white of course has no colour so they can't be there. 
The only error is as you say. skin.css
  "GET http://192.168.1.40/skin.css net::ERR_ABORTED 404 (Not Found)
loadSkinCSS @ (index):208
onLoad @ (index):208
(anonymous) @ (index):7"

There is a problem.  I use solid and then all I get is colour pallet not used. I do not believe it is a loading issue but not available.
Using b1 there is no problem, using b2 there is.

blazoncek, 2023-05-28:
You must have a configuration issue.
B2 introduced palette & color wheel hiding depending on effect & configuration used.

ceboxsell, 2023-05-28:
Sorry but no, you keep on finding excuses and not listening. It's fine you don't want to accept there is a problem. I have a work around that will get me by. There is a problem and it will come up for others so deal with it then. All good.  ---- On Sun, 28 May 2023 20:43:54 +1000  ***@***.***  wrote ----
You must have a configuration issue.
B2 introduced palette & color wheel hiding depending on effect & configuration used.

blazoncek, 2023-05-28:
If you want this issue to be resolved post your configuration (cfg.json) and your presets (presets.json).

Otherwise I am chasing ghosts.

ceboxsell, 2023-05-28:
I am happy to provide whatever you need.and to help. I totally understand that you need to find a basis for this but it feels like I am being dismissed. Anything you want I will get for you, will post config files shortly. I suspect the presets config will help .as I said a saved preset from before the change to enable 2d fixes the issue until a reboot. I will try and save the same config after the change and see if it shows anything for you.  ---- On Sun, 28 May 2023 22:29:13 +1000  ***@***.***  wrote ----
If you want this issue to be resolved post your configuration (cfg.json) and your presets (presets.json).
Otherwise I am chasing ghosts.

ceboxsell, 2023-05-28:
[wled_cfg.json.txt](https://github.com/Aircoookie/WLED/files/11584572/wled_cfg.json.txt)
[wled_presets.json.txt](https://github.com/Aircoookie/WLED/files/11584578/wled_presets.json.txt)
I did a factory reset and configured again before doing creating these.

The config file was changed only to formatted json and removed the SSID

blazoncek, 2023-05-28:
I see that you are using version from February. Please try to update to 0.14.0-b2 build 2305090 or later.

ceboxsell, 2023-05-28:
Exact steps.
1. Factory reset.
2. In Thony erase flash and instal micropython. 
3. In install.wled.me install V0.14-b1
4. in install.quinled.info/diguno install 0.14-b2
5. Set WIFI
6. Set pin to 2 and 256 LEDs
7. effect android set a pallet colour. test ok save preset
8. Set 2d panel. 32*8.
9. Test Akami is correct. wheel etc available.
10. Restart.  
11. Select any effect colour wheel and pallet gone.
12. select saved android profile and wheel and pallet returns for all where appropriate.

ceboxsell, 2023-05-28:
Thanks, I am happy to wait until new builds are available. just wanted you aware of what I have been seeing.  
Will do more tests with @srg74 's builds as I can't be arsed setting up to build myself at this time.

blazoncek, 2023-05-28:
Try a binary from [here](https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/v0.14.0-b2).

ceboxsell, 2023-05-28:
Thanks, used https://github.com/srg74/WLED-wemos-shield/blob/master/resources/Firmware/%40Aircoookie/v0.14.0-b2/universal_shield/WLED_0.14.0-b2_esp32_universal_shield_minimum_4mb.bin 
This was enough to test.  The issue is resolved in this version so was just an earlier build.
The 2d setup is much better now as well.

blazoncek, 2023-05-28:
Now you know you were not being dismissed. 😉 
BTW if you need audio, use another binary.