wled/WLED #3215: "Just-Read"-Mode for configurable GPIOs

SHU-red, 2023-05-23:
**Is your feature request related to a problem? Please describe.**
- In my case im using WLED flashed to a nodemcuv2 on my desk for ambient lighing
- I also want to additionally read some other stats --> In my case i want to use a button panel to control additional stuff
- I asked myself why to use additional microcontrollers for this input, if i already have my WLED controller with several unused GPIOs

**Describe the solution you'd like**
- There is already a section to easily use several GPIOs for WLED control in the settings
- These settings already support multiple input-types (Button, Switch, PIR, ...)
- Why not adding a checkbox for each configured GPIO to "Just-Read" the information but do not use it for switching WLED state
- This way i could e.g. use MQTT and read information from all available GPIOs without the need of custom building a WLED software

**Describe alternatives you've considered**
1. Set up a additional Microcontroller just to read buttons (Tasmota, openMQTTGateway) --> Additional Wifi-device, unused GPIOs on WLED & Button Controller etc.
2. Custom build WLED --> Additional effort, i do not have enough knowledge and dont know about follow-up problems like updating to a newer WLED version

**Additional context**
- I am not an advanced user, so there may already be a solution for this
- I dont' think (but i dont know), that this would not add much weight to the "standard-wled" software due to the fact that everything is already included (Sensor-Types, MQTT, ...)
- If i dont underestimate this feature request there must only be added checkboxes to the already available GPIO-configurations and the information has just to be sent via MQTT just in the same way as it already does (but without switching WLED stuff)

Hopefully i did express myself in a understandable manner and not miss somenthing already present in the docs

blazoncek, 2023-05-24:
Just assign preset that does nothing.

SHU-red, 2023-05-24:
Sorry at the moment im not able to test it quickly
I will try to do so in the coming weeks

Feel free to close this item if you think there is no misunderstanding and this "function" can already be used

blazoncek, 2023-05-27:
By creating a dummy preset that does nothing (e.g. `{"v":false}`) and assigning it to every action of a button you can have MQTT messages transmitted on button press but no action triggered in WLED.

closed 2023-05-27 by blazoncek