wled/WLED #3224: Will WLED ever support Luciferin ambilight?

makersteto, 2023-05-31:
Hello,
I'm an old WLED user but I'm currently using Luciferin as my main software to create an ambilight clone.

I think that Luciferin is the best in class ambilight clone software for PC but I haven't understood how to integrate it with WLED.

Is it possible to integrate Firefly Luciferin with WLED?
https://github.com/sblantipodi/firefly_luciferin

If the answer is no, 
do you consider supporting Luciferin in the future?

Thanks

Aircoookie, 2023-05-31:
Hi,
Luciferin sure looks like a great and easy to use screen capture software. As far as I can tell, it is currently not compatible with WLED, as it only outputs a custom UDP data format for the Glow Worm Luciferin firmware.

It would be able to interface with WLED if it supported output of a standard UDP LED stream format, like Art-Net, E1.31 or DDP. I would recommend [DDP](http://www.3waylabs.com/ddp/) as it is the fastest (least overhead) and easiest to implement.

It would be possible to make this work by writing a usermod that emulates the Glow Worm firmware, however the more flexible approach would be if DDP support was added to Firefly Luciferin (along with an option to specify device IP and type, or auto-discover with the `_wled._tcp` mDNS service)

In conclusion, I would recommend you open a Feature request on the Firefly Luciferin repo as well asking for WLED support.
Will leave this open, as writing a usermod is still an option if the Luciferin team does not want to add a different format to their software.

sblantipodi, 2023-06-02:
Hi, I have seen the "cloned" issue on the Luciferin repo and I answer here to update you all :)

As Aircoookie suggested it should be Luciferin to support WLED and not the opposite.
I'm obviously a big fan of WLED (I mean, who isn't?) and I already considered adding that support.

Glad to see Aircoookie kind words on Luciferin, much appreciated!

This kind of support it somewhat difficult for various reasons, some of the few that comes into my mind now are:

- Most of the Luciferin features can be accessed with and without WiFi while WLED is mostly a "connected" firmware. All the features are available via USB, HTTP, MQTT.
- Luciferin PC software has a complete control of the Luciferin firmware (GPIO configuration, LDR calibration, white temperature calibration, led number to use, mqtt params, ecc. ecc.), I doubt I would be able to control WLED firmware via the Luciferin PC software as deep as I can with the custom firmware.
- Luciferin PC software can auto update the companion firmware without user interaction.
- Luciferin firmware communicates with the PC instance giving it a lot of useful infos that are used by the PC software, for examples it shares the framerate that the microcontroller is able to sustain in that moment, if the microcontroller isn't able to sustain a certain framerate, pc software reduces the framerate to minimize frametime.
- Flexibility, there is no much sense for WLED in implementing specific features for Luciferin since WLED is much more "general purpose". Having a companion firmware increases the flexibility for the development of Luciferin and new features can be pushed much faster.

Thank you for the interest, I'll look into this more in depth to see what I can do.

blazoncek, 2023-06-02:
I used to be a control freak and wanted to control everything from one app/device. Reality is, you can't (unless you are a big fruit company).

As @Aircoookie mentioned DDP is easy to implement and adding it to your software will only benefit it. And users.

Of course, if you want to keep closed system to have control about every aspect of UX then unfortunately some users will be at loss.

pandaboy6621, 2024-10-19:
Was someone able to make a usermod?

sblantipodi, 2025-01-10:
@pandaboy6621 a usermod will not work, Firefly Luciferin PC software is deeply integrated with it's companion firmware.
a usermod will not be enough.