wled/WLED #3228: DDP More 800 LEDS

MatheusConti, 2023-06-02:
### What happened?

I am facing an issue with my ESP32. I have two boards, and each one has 1152 LEDs connected. I have 2 strips of 144 LEDs per pin on all 4 outputs, totaling 1152 LEDs. However, when I try to add the DDP for the other MCU with 1152 LEDs, it warns that it cannot have more than 800 LEDs per channel. However, each channel has 288 LEDs, not 800... is this a bug?

### To Reproduce Bug

When you try to add another MCU and the LED quantity is greater than 800.

### Expected Behavior

I believe it should calculate and recognize that the MCU has more than 800 LEDs, but they are not connected to just one pin.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MatheusConti, 2023-06-03:
[image]
What he doesn't understand is that these 800 LEDs are divided into 4 outputs.

blazoncek, 2023-06-03:
As far as master unit goes it sees slave unit as a single continuous strip.
There is nothing that can be done about that.

And BTW maximum recommended length for slave unit is 480 pixels.
Closing.

closed 2023-06-03 by blazoncek