wled/WLED #3229: Wifi connect is not possible ESP32 S2 (loop on AP)

chryss44, 2023-06-03:
### What happened?

When i complet all wifi connection informations, my ESP restart and stay on AP.
If i reset it, it loop on AP access without try to connect at my wifi

### To Reproduce Bug

[Video-Sat-Jun-03-2023-12-16-49.webm](https://github.com/Aircoookie/WLED/assets/32177747/27e45aa6-eaff-408f-aa6b-3bf5739b19c1)

### Expected Behavior

I just want to use ESP32 S2 on my wifi, not on the AP.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b2 (build 2306020)

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Anything else?

I have try my version with Wemos 8266 and everything is OK

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-06-03:
Actually I have an -S2 here that does connect to the wifi after entering the wifi password in AP mode. So it's not a generic problem.

Please use WLED [Discord](https://discord.gg/KuqP7NE) for help and support questions.

Some generic ideas:
* make sure your wifi runs on 2.4Ghz - esp32 cannot connect to 5Ghz.
* double check the password
* support for ESP32-S3 is still experimental. Maybe your board simply does not work well with WLED.

blazoncek, 2023-06-03:
I have now about 6 ESP32-S2 devices all nicely configured and running. No issues whatsoever.
If you are compiling yourself, try adding:
```
  -DARDUINO_USB_MSC_ON_BOOT=0
  -DARDUINO_USB_DFU_ON_BOOT=0
  -DLOLIN_WIFI_FIX
```

closed 2023-06-03 by blazoncek

dkebler, 2024-01-07:
so were those added to the latest s2 pre release as I have no success seeing AP mode after flashing.

https://discord.com/channels/473448917040758787/473449548279185408/1193375929280516127

dkebler, 2024-01-07:
I had an s2  "nodemcu style" devkit and I can flash that via the web installer. and I can see the AP but it won't connect.

just for grins I flashed a regular esp32 devkit and no problems.  Can connect to AP and get on my network.   **This is an esp32s2 issue.**

dkebler, 2024-01-07:
rather than necro this closed issue maybe I should open another one?  Despite @blazoncek  experience this is still a bug for me 6 months later.   See my discord post.

blazoncek, 2024-01-07:
@dkebler unfortunately you'll need to research what goes wrong with your S2 yourself by monitoring serial output and using custom debug builds.

I can only say that with the current PIO environments Lolin ESP32-S2 in D1 mini form with 2MB of PSRAM works flawlessly. It may have WiFi connectivity problem when signal is very weak or you are using advanced WiFi features but otherwise opens AP successfully and connects to a 2.4GHz network w/o issues.

If you are unable to make your particular board work, throw it away and get Lolin board. They are by far the cheapest I found, costing less than comparable ESP8266.