wled/WLED #3230: Feature-Request: API command to remap buttons to new destination ID with JSON API / within Preset

nouser2013, 2023-06-05:
**Is your feature request related to a problem? Please describe.**
I may have missed it, but one thing missing in the extensive API is imho the option to remap buttons upon execution of a preset or with an API call.

**Describe the solution you'd like**
What I'd like to see is the option to remap `short-press()`, `long-press()` and `double-click()` to new preset or playlist IDs so that I can be more flexible in structuring the button navigation. Ideally, such a command would take three arguments:

1. Button Number
2. Button Press mode (short, long, double)
3. New destination preset / playlist ID

**Describe alternatives you've considered**
I tried using the MIN/MAX preset-ID and the cycle function, but for my use case it is not flexible enough.

blazoncek, 2023-06-05:
You can configure buttons in the settings (Time & Macros).

nouser2013, 2023-06-05:
Ah thanks, I get that. But I would like to re-assign the buttons depending on the preset / playlist being active at the moment:

* (1) MAIN - Static
* (2) BL - Animation
* (3) BR - Animation
* (4) BLUE - Animation
* (5) YELLOW - Animation
* (6) BL5s - Playlist Once
* (7) BR5s - Playlist Once

Desired state chart (S=single, D=double, L=long press, B0=button0, B1=button1):
[image]

It's a bit patchy, but I hope one can get the idea.

blazoncek, 2023-06-05:
The description and image are meaningless to me.

nouser2013, 2023-06-05:
hmmm ok, I'll try a different explanation.

* if ESP boots, I

blazoncek, 2023-06-05:
A little better.
Why don't you just use multiple buttons? You can have up to 16.

nouser2013, 2023-06-05:
Yes, I could do that. Two drawbacks that I see:
* Buttons => Hardware: more effort for wiring, higher costs, etc.
* I cannot use the ready-made image from the web installer anymore, I'll always have to build myself (as far as I know).

Regarding effort: Well, something happens when you press "save" in the settings.
There would "just" a new entry in the command processor for presets be needed which triggers the same logic.

But, alas, I'm not a developer, so I cannot truly know the real effort.

blazoncek, 2023-06-06:
@Aircoookie would you consider adding configuration changes to JSON API?

Something in the lines of:
```c++
  // accepting config changes (not permanent, unless config is saved)
  JsonObject cfg = root["cfg"];
  if (!cfg.isNull()) {
    // reconfigure button presets (on the fly)
    JsonVariant btn = cfg["btn"];
    if (btn.is<JsonObject>()) {
      uint8_t btn_id = btn["id"] | 0;
      macroButton[btn_id]      = btn["s"] | macroButton[btn_id];
      macroLongPress[btn_id]   = btn["l"] | macroLongPress[btn_id];
      macroDoublePress[btn_id] = btn["d"] | macroDoublePress[btn_id];
    } else {
      uint8_t btn_id = 0;
      for (JsonObject _btn : btn.as<JsonArray>()) {
        btn_id = btn["id"] | btn_id;
        if (btn_id >= WLED_MAX_BUTTONS) continue;
        macroButton[btn_id]      = _btn["s"] | macroButton[btn_id];
        macroLongPress[btn_id]   = _btn["l"] | macroLongPress[btn_id];
        macroDoublePress[btn_id] = _btn["d"] | macroDoublePress[btn_id];
        btn_id++;
      }
    }
  }
```

Aircoookie, 2023-06-06:
@blazoncek we already have that, `/json/cfg`, including `{"sv":false}` if immediate write to FS cfg.json is not desired.

In order to change config from within a preset, this would of course need to be accessible from within `/json/state` as well.
Not a fan of duplicating config deserialization code though, it seems hard to maintain and we'll get requests to make setting XY available via presets as well, so could be best to just allow changing any setting. Something like this would be acceptable IMO:

```cpp
JsonObject cfg = root["cfg"];
if (!cfg.isNull()) { 
   if (!cfg["sv"].isNull()) doSerializeConfig = cfg["sv"]; // async save to FS if explicitely requested
  cfg["sv"] = false;
  deserializeConfig(cfg);
}
```

Would need to be tested, I can't immediately judge whether changing cfg from presets (and potentially async saving to FS) could have a security or stability impact. One risk factor may be someone copy-pasting an unknown preset json that messes up their config without their knowledge.

blazoncek, 2023-06-07:
@Aircoookie I know about `/json/cfg` but that is not part of the regular preset parsing (as you noted). The other issue with `/json/cfg` is the complexity of configuration items which may be unnecessary to expose to presets (as to reduce their size) and the lack of user friendly keys (you would need extensive documentation which is lacking ATM).
Buttons, for example, would need nested arrays with all items present to correctly override behaviour.

There is also a security issue when using `/json/cfg` as it does not respect PIN settings. Allowing you to save configuration regardless of PIN protection.

I am also not fan of duplicated processing and though intention of `/json/cfg` was good it brought its issues with it as it does not conform to regular configuration workflow which I am not very fond of.

blazoncek, 2023-06-13:
Until this is agreed upon between collaborators here is the code that will help you remap buttons _on the fly_.
```c++
  // accepting config changes (not permanent, unless config is saved)
  JsonObject cfg = root["cfg"];
  if (!cfg.isNull()) {
    // reconfigure button presets (on the fly)
    JsonVariant btn = cfg["btn"];
    if (btn.is<JsonObject>()) {
      uint8_t btn_id = btn["id"] | 0;
      macroButton[btn_id]      = btn["s"] | macroButton[btn_id];
      macroLongPress[btn_id]   = btn["l"] | macroLongPress[btn_id];
      macroDoublePress[btn_id] = btn["d"] | macroDoublePress[btn_id];
    } else {
      uint8_t btn_id = 0;
      for (JsonObject _btn : btn.as<JsonArray>()) {
        btn_id = _btn["id"] | btn_id;
        if (btn_id >= WLED_MAX_BUTTONS) continue;
        macroButton[btn_id]      = _btn["s"] | macroButton[btn_id];
        macroLongPress[btn_id]   = _btn["l"] | macroLongPress[btn_id];
        macroDoublePress[btn_id] = _btn["d"] | macroDoublePress[btn_id];
        btn_id++;
      }
    }
  }
```

Add them to json.cpp just above `stateUpdated(callMode);` arround line 458.

nouser2013, 2023-06-14:
Oh marvellous, thanks a lot. I'll be sure to give this a try.

Also, I'd still would wish for this feature to appear in the `main` branch at some point and would request that this issue be kept open.

blazoncek, 2023-06-14:
Perhaps...
I hope we've earned a beer until then. 😉