wled/WLED #3235: Usermod BME280 not showing

AxelHoss, 2023-06-07:
### What happened?

When i create my own platformio_override.ini with d1_mini and the bme280 section it wont show me a usermod in the setup page on the esp8266. I dont geht any error on compiling or upload.

`[platformio]
default_envs = d1_mini

[env:d1_mini]
board = d1_mini
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
upload_speed = 921600
board_build.ldscript = ${common.ldscript_4m1m}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266}
lib_deps = ${esp8266.lib_deps}
monitor_filters = esp8266_exception_decoder

[env:usermod_bme280_d1_mini]
extends = env:d1_mini
build_flags =
  ${common.build_flags_esp8266}
  -D USERMOD_BME280
lib_deps = 
  ${esp8266.lib_deps}
  BME280@~3.0.0
  Wire`

[image]

### To Reproduce Bug

see abve

### Expected Behavior

i want to see the usermod on the wled homepage

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-B2 Hoshi

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-06-07:
Please contact original author (use Git blame).
If you cannot see BME280 in Info dialog then it was not compiled into binary BTW.

blazoncek, 2023-09-18:
No response received, closing.

closed 2023-09-18 by blazoncek