wled/WLED #3236: suspicious code in e131.cpp (detect RGB / RGBW)

softhack007, 2023-06-07:
This line is a bit suspicious, and I doubt it will work as intended:

https://github.com/Aircoookie/WLED/blob/999bec19f1bf37b344177a9aeffc752c5f4ae472/wled00/e131.cpp#L28

Compiler warning: 
````
wled00/e131.cpp:29:57: warning: suggest parentheses around comparison in operand of '&' [-Wparentheses]
   uint8_t ddpChannelsPerLed = (p->dataType & 0b00111000 == 0b011) ? 4 : 3; // data type 0x1B (formerly 0x1A) is RGBW (type 3, 8 bit/channel)
                                              ~~~~~~~~~~~^~~~~~~~
````
If it is intended as `((p->dataType & 0b00111000) == 0b011)`, then the decision cannot become true. Any value & 0b00111000 will not have the lowest 3 bits sets, as they are masked out by "&".

Actually I'm not sure how it was indented to work, maybe the original author could take a look?

blazoncek, 2023-06-07:
It has been fixed in `beta-3` branch.

Aircoookie, 2023-06-07:
Embarrassing 😅 Well, mistakes happen, thanks for catching it @softhack007 and fixing it promptly @blazoncek :)

closed 2023-06-16 by Aircoookie