wled/WLED #3240: On power ON WLED load preset #0 first, and not the configured one

Egor-Li, 2023-06-09:
### What happened?

Hello,
I have a quinled-dig-uno board with WLED flashed. My LED strip is W2815. The WLED is configured to load preset #1 at startup, which is set to white color. 
When the controller receives power, it first lights up red for half a second, and only then loads preset #1, which is white.
Here's a slowmo recording: https://www.youtube.com/shorts/rgeKIuFiTtA
The issue seems that there is a delay in reading set default preset. WLED loads and immediately applies the default preset #0, then reads from the config that the default preset should be #1 - then applies the #1 preset. 

### To Reproduce Bug

Set the startup preset to anything but #0 and different color and apply power to the controller with WLED. Before running the desired preset, it will flash red.

### Expected Behavior

Launch the correct configured preset, without reading preset #0 first.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-06-09:
Already fixed.
Duplicate of #3219

closed 2023-06-09 by blazoncek

Egor-Li, 2023-06-10:
Not fixed on my end, unfortunatelly.
I have two quinled dig uno controllers. On one of them I saved the preset without brightness, and also turned off LED power on on power reset. Now it doesn't exhibit this behaviour. Even though LED ON on power reset is disabled, the LED strip still lights on.
My second unit with the same firmware responds a bit more expectedly - if LED on on power reset is OFF, the LEDs do not light up when power is reset. However, saving the preset without brightness settings doesn't fix the issue as with the first strip - the color is still flickering on power reset.

blazoncek, 2023-06-10:
There is no preset 0.
Update to latest nightly build of WLED where the bug is fixed.

Egor-Li, 2023-06-10:
> There is no preset 0. Update to latest nightly build of WLED where the bug is fixed.

Oh, that was an oversight on my end. The latest beta version in releases is 0.14.0-b1. Somehow found a 0.14.0-b2 through google - indeed the issue is fixed. Thank you!