wled/WLED #3245: RGB Segment DMX-Mode (Effect Segment but Realtime and only RGB channels)

maxnuglisch, 2023-06-12:
I build a logo sign for my concert show that i would like to be controlled by DMX without the need of a pixel mapper like Madrix.

The Effect Segment mode would fit my needs but it is far from realtime which basically makes it unusable for controlling it live.

I think a usefull and logical addition would be a RGB Segment mode, which is basically like the Effect Segment mode but with only RGB channels per segment and its realtime. (maybe also add a DRGBW Segment version as well) 

To make it clear: Yes. You can achieve this funcionality by using the Effect Segment mode and just using the solid effect. But thats far from realtime :( So a more minimal but faster approach would be a perfect addition!

I think that would make using wled devices as DMX fixtures a lot more easy and I would have a lot of other ideas on how to use this functionality :)

Really love the Project! <3

mxklb, 2023-08-14:
Have you tried @blazoncek workaround mentioned [here](https://github.com/Aircoookie/WLED/issues/3239#issuecomment-1673954039)?

I use WLED for live DMX control as well. It's slow for some effects/channels, but others seem to work better ..

deggle, 2023-10-21:
I asked for exactly this some time ago, and for these reasons - for instances where segments are clearly defined it would be great to have a basic RGB(W) function (potentially with a Dimmer channel per segment, and overall) for operation from a lighting desk. I even put a bounty on this as I felt it would be a useful (and I believe) quite simple implementation.

https://github.com/Aircoookie/WLED/issues/2325#issuecomment-1133602023

However, in my own setup I'm tending to use the Effect Segment now as for some occasions being able to use WLED effects is useful. I've things like LED rings and to run effects from a lighting desk would then need individual LED addressing and the resulting 100s of channels that I wanted to avoid.

I've hit the same problem that even just flashing causes WLED to do a fade in and fade out and it's hugely laggy. But - I have noted the workaround mentioned above, and am having another play now! Given I do find the effects useful, getting live/low latency RGB too would be the best of both worlds for me. But I do still think (D)RGB(W) segments could be useful and save a lot of channels if the FX aren't needed.

deggle, 2023-10-21:
I've been having a play with this, and I'm not finding much improvement. I disabled the crossfade which has helped a little, but RGB changes are very laggy. Oddly, I can quite rapidly flash an FX with all it's settings (palette, speed ete) and it will respond pretty much instantly. But if I flash a RGB (Speed 0, palette 1 Col etc) it's laggy. This is odd as surely setting up a fully animated FX takes much more computational than rendering a static RGB. Odd.

deggle, 2023-10-22:
I had a couple of replies on here, but I've deleted them as this is essentially a duplication of https://github.com/Aircoookie/WLED/issues/3239 which has a workaround solution.