wled/WLED #3250: Peek fails to load if websockets connection can't be established

blazoncek, 2023-06-14:
### What happened?

If WLED is compiled with Websockets the Peek LED output preview fails to load if websocket connection can't be established for some reason (proxy, firewall, etc). It should gracefully fall-back to non Websocket implementation instead.

### To Reproduce Bug

Add a firewall or a reverse proxy which will block websockets between WLED and client.

### Expected Behavior

Peek should work with or without Websockes, regardless of the way how WLED was compiled (with or without Websockets).

### Install Method

Self-Compiled

### What version of WLED?

0.14.0-b3 2206130

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32, ESP32-S2, ESP32-C3

### Relevant log/trace output

```shell
N/A
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2023-06-21:
So if you compile without there is an alternative implementation already? Long polling?
If you make the appropriate non-ws call to a WS enabled build does it work? I.e is it just the front-end code that would need to change?

blazoncek, 2023-06-21:
Both will need an update.
And yes, if you disable websockets, peek works without.

netmindz, 2023-06-21:
Not sure how to best detect the failure, I've seen poorly written firewalls allow the websocket to be negotiated, but then block any actual messages, so protocol switch is not always enough to prove success

blazoncek, 2023-06-21:
IMO fairly easy. When WS is not established or times out fallback to HTTP.
Easy in GUI more work in backend.

netmindz, 2023-06-21:
My point is that sometimes it can report as established, but isn't really, so would require timeout for an actual message, not just the http and protocol switch

blazoncek, 2023-07-03:
Partially solved by https://github.com/Aircoookie/WLED/commit/f015227fc8aa34cd511d8b1c9d91cd0a5c5c9067
Still missing 2D peek.

blazoncek, 2023-07-18:
I have asked @ewoudwijma to look into implementing 2D peek using HTTP JSON requests.

blazoncek, 2023-08-30:
Not going to pursue this any further.

closed 2023-08-30 by blazoncek