wled/WLED #3252: AP not working on esp32c3dev.bin (c04c73b)

Titanium177, 2023-06-18:
### What happened?

AP comes up, but is not connectable with newest esp32c3dev.bin

### To Reproduce Bug

Flash newest esp32c3dev.bin to ESP32-C3 from https://github.com/Aircoookie/WLED/actions/runs/5300577446

### Expected Behavior

Expected normal setup routine (connect to AP & fill in WIFI settings)

### Install Method

Binary from WLED.me

### What version of WLED?

see "To Reproduce Bug"

### Which microcontroller/board are you seeing the problem on?

ESP32-C3

### Anything else?

Board: https://www.electrodragon.com/product/esp-led-strip-board/

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-06-18:
ESP32-C3 (and S2 or S3) are still in experimental phase.
You will need to tweak build environment to accommodate your particular dev board.
For a few examples look into `beta-testing` or `dev-general` topics on Discord.

Titanium177, 2023-06-21:
Tried to debug, but it shows literally nothing... just resets. Maybe you see more than me. (https://github.com/Aircoookie/WLED/files/11811353/LOG.txt)

Also found a working firmware, which just don't include Alexa. Is it possible to get the settings from that bin and just add Alexa? (https://github.com/Aircoookie/WLED/files/11811364/C3_firmware.zip)

dosipod, 2023-06-21:
Hi , Point to keep in mind is what blazoncek said about the new  chips being in experimental phase. , even though  the  c3 i tested with you worked before and seems stable  when not  messed with too much , i am also only experimenting with that so if your aim is a production setup at this time then please reconsider . 
If the aim is  experimenting  to try to improve the support for new chips  then by all means ( we might be able to help  compile and test with  you ) well with the limitation i mentioned in discord that I only have one c3  so I can not do double units tests but might be you have more then one

Titanium177, 2023-06-22:
> Hi , Point to keep in mind is what blazoncek said about the new chips being in experimental phase. , even though the c3 i tested with you worked before and seems stable when not messed with too much [...]

Yes, you guessed it. I'm aiming to get a prod-env, to use them in my smart home with IoB. Since there's none yet, I'm fine with testing/experimenting to get them there. 

I'm also not new to coding in general. If I get an "error" in log, I'm able to work with that. Since the log above shows none, just resetting, I can't work with that.

Btw. got 20 of these Boards, so if you explain me what you exactly mean with "double units tests", I should be able to do that.

closed 2023-06-30 by Titanium177