wled/WLED #3255: problem compiling wled

Francesco-Rugg, 2023-06-21:
### What happened?

I'm trying to compile wled because I would like to add the display usermod, but I always have problems.
I was thinking that the problem was my platformio_override.ini but even if I try to do a simple compile I got always the same error. With or without the user mode. The error is     `SyntaxError: Unexpected token ‘<’,"`

this is my override file:

`[platformio]
default_envs = d1_mini

[env:d1_mini]
board = d1_mini
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
upload_speed = 921600
board_build.ldscript = ${common.ldscript_4m1m}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266}
    -D USERMOD_FOUR_LINE_DISPLAY -D FLD_PIN_SCL=5 -D FLD_PIN_SDA=4
    -D LEDPIN=2 -D BTNPIN=0
lib_deps = ${esp8266.lib_deps}
    U8g2@~2.34.22
    Wire
monitor_filters = esp8266_exception_decoder`

Now if I want to do another time a build VS code doesn't generate the build output file anymore... And I always need to reinstal VS code
what I have to do?

### To Reproduce Bug

try to compile with VS code on windows

### Expected Behavior

Use wled with display usermod

### Install Method

Self-Compiled

### What version of WLED?

WLED beta release 0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-06-21:
PIO  has been reported  to  misbehave   due to updates  or lack of it , this time might be best to compile , still do that and do not give up , meantime why not test the pre-complied bins from here https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/v0.14.0-b3

Francesco-Rugg, 2023-06-21:
> PIO has been reported to misbehave due to updates or lack of it , this time might not be best to compile , still do that and do not give up , meantime why not test the per-complied bins from here https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/%40Aircoookie/v0.14.0-b3

Thank you for the link but same issue. when i use the bin from version 0.14.0-b3 I always get the same syntaxerror but with version 0.13 I don't have this error and everythings works. but why?

Francesco-Rugg, 2023-06-21:
Now I'm trying to compile in Visual studio version 0.13 but always got this error (same with version 0.14 downloaded from github soure code)

[image]

this problem appear only when I use in visual studio the source code, and not when I use wled-main.

softhack007, 2023-06-21:
> this problem appear only when I use in visual studio the source code, and not when I use wled-main.

Older platformIO.ini have some buildenv names with "." (dot), but the current platformIO version does not like that any more.

Try to rename the build envs similar to this patch:
https://github.com/Aircoookie/WLED/commit/bc90309dd69533ff9f4b879fb82ecfd1b1db35d7

Bacially search for `codm-controller-0.6` and replace with `codm-controller-0_6` .

dosipod, 2023-06-21:
> Thank you for the link but same issue. when i use the bin from version 0.14.0-b3 I always get the same syntaxerror but with version 0.13 I don't have this error and everythings works. but why?

I want to try to replicate , so you are having issue with latest 14 b3 with and without  ( USERMOD_FOUR_LINE_DISPLAY  )  ? do you see issue  with esp8266 without any usermods ?

closed 2023-06-21 by blazoncek