wled/WLED #3256: Using Mbi6033 with WLED (Blinking and Empty Data problem)

seyfettinkazanir, 2023-06-21:
I want to control led strips thanks to WLED with mbi6033, a chip that was added to the NPB library some time ago.

It is a chip that uses mbi6033 dta+clk, added to the neopixelbus library. I’ve come to a certain point by making some changes to the wled code. But I can’t solve flickering and empty color bytes.

mbi6033 asks for header information first and then uses 48 bit color. In the form of G 16 + B16 + R16. However, the first large byte of this 16-bit is always empty. I added in the picture.

The method we used in the bus_wrapper section was added this way.
`#define B_HS_DOT_3 NeoPixelBusLg<NeoRbg48Feature, Mbi6033Method, NeoGammaNullMethod> `

The data shown in the image below represent: Color = White , brightness = full (from wled control panel)
[image]

The data shown in the image below represent:  single color, full brightness
[image]

blazoncek, 2023-06-21:
*Some time* is 2 days as of this writing. 😀  It will not make it into WLED until new NeoPixelBus is released.
And even then it may take time. You may accelerate the process by sending sample LED strip.

seyfettinkazanir, 2023-06-21:
Are you talking about sending a physical led strip?

blazoncek, 2023-06-21:
yes

seyfettinkazanir, 2023-06-21:
Where do I need to send it, can you send an address?

seyfettinkazanir, 2023-06-23:
@blazoncek I want to send led strip prepared with mbi6033, how should I do?

blazoncek, 2023-06-23:
I am currently too busy to tackle this implementation. Perhaps @softhack007 or @Aircoookie are willing to take on the challenge?

If not, you are still free to implement additional LED chip support in your fork and make a PR upstream. You will need to update (at least) the following files:
- settings_leds.htm
- bus_wrapper.h
- const.h

As a base take UCS890x support.

seyfettinkazanir, 2023-06-23:
I would be very grateful if anyone can help.  I may need all the help. I can send in physical led strip if needed.

settings_leds.htm
bus_wrapper.h
const.h

I tried to make the edits you mentioned. It's a little dim but I couldn't figure out the flickering. I edited it a bit with the mbi6033method prepared in the NPB library. Would it be helpful in terms of help if I stated the edits I made here?

softhack007, 2023-06-23:
I could take it, but not this month - mid July would be earliest possible for me to start thinking about a solution.

Actually there is a known bug in NeoPixelBus regarding 16bit byte ordering ("endianness") that is only solved in the dev branch of makuna. So it would be smart to wait until a new NPB release is availeable.

* https://github.com/Makuna/NeoPixelBus/issues/700

softhack007, 2023-06-23:
> couldn't figure out the flickering.

I've seen several people reporting unexpected behaviour of the brightness limiter in latest WLED. Maybe this also causes the flickering you experience?

seyfettinkazanir, 2023-07-03:
> I could take it, but not this month - mid July would be earliest possible for me to start thinking about a solution.
> 
> Actually there is a known bug in NeoPixelBus regarding 16bit byte ordering ("endianness") that is only solved in the dev branch of makuna. So it would be smart to wait until a new NPB release is availeable.
> 
> * [ColorFeatures using 16bit color elements has endian issue with buffer copies to color objects Makuna/NeoPixelBus#700](https://github.com/Makuna/NeoPixelBus/issues/700)

I guess it looks like a new NPB version is available.  https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.6 

Would you have time to add the Mbi6033 if it won't tire you out?

blazoncek, 2023-07-03:
> Would you have time to add the Mbi6033 if it won't tire you out?

Last NPB update caused headaches we are still recovering from. :)

seyfettinkazanir, 2023-07-08:
@blazoncek  Still developing headaches with NPB latest update? I wanted to ask if it's the right time to work on the mbi6033.

blazoncek, 2023-07-08:
Let's say it is scheduled for 0.15 release. Ok?
😄

seyfettinkazanir, 2023-07-08:
> 

Thanks, I'm looking forward to it.

seyfettinkazanir, 2023-08-05:
> Let's say it is scheduled for 0.15 release. Ok? 😄

Greetings, have you had the opportunity to do any work on the Mbi6033? This product excites me a lot. I'm looking forward to your news. Could a schedule be specified for 0.15 release?

softhack007, 2023-08-05:
> > Let's say it is scheduled for 0.15 release. Ok? 😄
> 
> Could a schedule be specified for 0.15 release?

Considering that all developers are doing it in their spare time (hobby), it's very hard to give any dates. 😉 "it's ready once we consider it well-done".

I think all we can say today is
* 0.14.0 is still bet and not ready to be released
* 0.15.0 will come after 0.14.x

seyfettinkazanir, 2024-03-21:
Hello, has there been any progress regarding mbi6033 for 15.0? We look forward to any developments in this regard.

@softhack007 @blazoncek

blazoncek, 2024-03-21:
No. But you can submit PR. 0.15 as of now uses NePixelBus 2.7.8 which is the latest release available ATM.