wled/WLED #3257: Update failed with WLED_0.14.0-b3

javigmsuperka, 2023-06-22:
### What happened?

Update failed on athom LS4P-WLED devices:

https://es.athom.tech/blank-1/wled-high-power-led-strip-controller

[image]

Used firmware WLED_0.14.0-b3_ESP02.bin / esp8266  with 2mb flash.

### To Reproduce Bug

Install last beta firmware on athom device.

### Expected Behavior

Failed to update.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED_0.14.0-b3

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

olympia, 2023-06-22:
I've got the same both with:
 - https://www.athom.tech/blank-1/wled-2812b-led-strip-controller
 - https://es.athom.tech/blank-1/wled-high-power-led-strip-controller

Aircoookie, 2023-06-22:
Thank you for the report! Which version did you try updating from? Please check the total size of the Filesystem in the Info tab of the main UI, it should be 499 kB.

I successfully OTA updated my 7W E27 Athom bulb, which also runs on a 2MB ESP8266.

javigmsuperka, 2023-06-22:
> Thank you for the report! Which version did you try updating from? Please check the total size of the Filesystem in the Info tab of the main UI, it should be 499 kB.
> 
> I successfully OTA updated my 7W E27 Athom bulb, which also runs on a 2MB ESP8266.

Hello!

[image]

olympia, 2023-06-22:
@javigmsuperka was faster, but it is exactly the same for me

weppa, 2023-06-22:
> I've got the same both with:
> 
>  - https://www.athom.tech/blank-1/wled-2812b-led-strip-controller
> 
>  - https://es.athom.tech/blank-1/wled-high-power-led-strip-controller

Same with me, I have the https://es.athom.tech/blank-1/wled-high-power-led-strip-controller

Toggz, 2023-06-22:
Problem here also using an Athom esp8266, manual and auto update via Ha failures. :(

Joakim-T, 2023-06-22:
Same issue here. Can i enable some kind of debugging in wled to see where it fails?

Rushmed, 2023-06-22:
Same issue here after updating ESP8266 WLED_0.14.0-b1_ESP8266.bin to WLED_0.14.0-b3_ESP8266.bin.
Tried OTA and wired flashing.
ESP stays in AP-Mode. Its controllable but does not connect to wifi.

blazoncek, 2023-06-22:
Are you using newer or older Athom bulbs?

Joakim-T, 2023-06-22:
Tried updating two of these Athom LS4P-WLED devies not with same result.
Tried manual update through the WebUI with the file WLED_0.14.0-b3_ESP02.bin.
Tried manual update and unchecking Enable ArduinoOTA (Probably unrelated).
Both fail with the "Please check your file" error.

Also tried updating using the Home Assistant integration and that fails with the following information in the logs...

Logger: homeassistant.core
Source: components/wled/helpers.py:34
First occurred: 07:21:24 (2 occurrences)
Last logged: 14:12:49

Error executing service: <ServiceCall update.install (c:01H3GRRVN95D8KB6QYV0C97X0J): entity_id=['update.arbetsrum_wled_list_firmware'], version=0.14.0-b3, backup=False>
Error executing service: <ServiceCall update.install (c:01H3HGAPGRVE43HY05X1RVW99H): entity_id=['update.ellie_wled_list_firmware'], version=0.14.0-b3, backup=False>
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 699, in upgrade
    await self.session.post(url, data=form)
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client.py", line 560, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.11/site-packages/aiohttp/client_reqrep.py", line 899, in start
    message, payload = await protocol.read()  # type: ignore[union-attr]
                       ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/aiohttp/streams.py", line 616, in read
    await self._waiter
aiohttp.client_exceptions.ClientOSError: [Errno 32] Broken pipe

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/wled/helpers.py", line 28, in handler
    await func(self, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/wled/update.py", line 89, in async_install
    await self.coordinator.wled.upgrade(version=version)
  File "/usr/local/lib/python3.11/site-packages/wled/wled.py", line 717, in upgrade
    raise WLEDConnectionError(msg) from exception
wled.exceptions.WLEDConnectionError: Timeout occurred while communicating with GitHub for WLED version information

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1928, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1950, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 226, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 811, in entity_service_call
    future.result()  # pop exception if have
    ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1034, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 851, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 156, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 417, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/wled/helpers.py", line 34, in handler
    raise HomeAssistantError("Error communicating with WLED API") from error
homeassistant.exceptions.HomeAssistantError: Error communicating with WLED API

mateuszdrab, 2023-06-22:
Same here, just got my first Athom device and was about to start experimenting with WLED and an LED matrix but b3 can't be flashed, b1 works fine. Great work on 2D support though, works pretty well.

mikem30303, 2023-06-22:
athom controller, same issue. Following.

softhack007, 2023-06-22:
Just an idea - the new firmware is 823Kb, while beta-1 was 784 Kb. Depending on LittleFS size, it could be that the firmware now is too large for normal OTA update, as OTA seems to need 2x firmware size. 2x823 =1646, so a 2MB devi ce with 512MB filesystem would not have enough space for OTA.

gvulvae, 2023-06-23:
Same issue. Following

riddlenl, 2023-06-24:
Indeed unable to update Athom WLED boxes. I was able to update my ówn ESP-8266.

StephanBruh, 2023-06-24:
Same issue with my Athom controller, can confirm.

Aircoookie, 2023-06-24:
> Just an idea - the new firmware is 823Kb, while beta-1 was 784 Kb. Depending on LittleFS size, it could be that the firmware now is too large for normal OTA update, as OTA seems to need 2x firmware size. 2x823 =1646, so a 2MB devi ce with 512KB filesystem would not have enough space for OTA.

This is it! Seems like we will need to disable some functionality in the default build to restore updatability. I was able to update my bulb as it ran 0.13.0-b5 or some other ancient version, the size of which thus must have been lower than `1500 - 823`, but there is of course not enough space.

This puts us in a tough space, as we need to ensure consistant binary sizes below 750kB to ensure updatability of 2M devices with 500kB FS. In order for an update from b1 to work, the new binary would even have to be less than about 716 kB, which seems quite hard to achieve. Probably will need to resort to a two-stage update process with a "minimal" binary version, I know that Tasmota has/had such a process to enable OTA capability even on 1MB flash devices.

softhack007, 2023-06-24:
> resort to a two-stage update process
Yes probably the best way forward for 8266 devices.

> This puts us in a tough space, as we need to ensure consistant binary sizes below 750kB to ensure updatability of 2M devices with 500kB FS.

In theorie we could also reduce FS size to 256 KB, which would allow for up to 875K images and normal OTA. The only downside would be that user will lose FS content when changing from 512KB to 256KB for the first time.

mateuszdrab, 2023-06-24:
> Yes probably the best way forward for 8266 devices.
>
>
> In theorie we could also reduce FS size to 256 KB, which would allow for up to 875K images and normal OTA. The only downside would be that user [...]

Wouldn't that mean a physical reflash of the device is needed. I doubt partition size can change with OTA, right?

Or maybe Tasmota style builds with feature variants, e.g. in my case support for audio isn't needed so a build without that would save lots of space. I'm guessing the same would apply to those who don't need 2d matrix support.

Aircoookie, 2023-06-26:
> Using the compressed binaries should update successfully

This is a great tip and I can confirm this to be working! It slightly increased update time as decompression is required, but I've added a new binary `[WLED_0.14.0-b3_ESP02_2306240.bin.gz](https://github.com/Aircoookie/WLED/releases/download/v0.14.0-b3/WLED_0.14.0-b3_ESP02_2306240.bin.gz)` to the release.

We still need to keep an eye on binary size, as the sum of b3 and b3 .gz combined is 1447, which is only about 50 kB below limit.

olympia, 2023-06-26:
I confirm this to work! Thank you!

WarC0zes, 2023-06-26:
Thanks for the tip, it works. 👍

javigmsuperka, 2023-06-26:
> This is a great tip and I can confirm this to be working! It slightly increased update time as decompression is required, but I've added a new binary [...]

Working.

Thanks!

Joakim-T, 2023-06-26:
Works! Thanks for a quick and simple fix 🙂

gvulvae, 2023-06-26:
Works fine! Thanks!

closed 2023-06-27 by Aircoookie

brentcomeau, 2023-08-02:
This fix worked for my newer Athom bulbs upgrading to 0.14.0-b3 from 0.13.3, but not my older bulbs running 0.13.0-b6.

danielbrunt57, 2023-08-07:
This compressed binary worked for my LS-4P from b1 to b3!  Thanks!

Balooski, 2023-08-17:
Same issue on latest firmware. Compressed binary did not work for me. Can I help in any way?

brentcomeau, 2023-10-09:
> This is a great tip and I can confirm this to be working! It slightly increased update time as decompression is required, but I've added a new binary [...]

This compressed file (and WLED_0.14.0-b6_ESP02.bin.gz) and still not working for athom bulbs stuck on 0.13.0-b6.

N1029676, 2023-11-22:
This process was failing when I tried to update from my phones web browser but was successful when updating from my desktop's browser.

Balooski, 2023-11-22:
I'm on 1

> > > Using the compressed binaries should update successfully
> > 
> > 
> > This is a great tip and I can confirm this to be working! It slightly increased update time as decompression is required, but I've added a new binary `[WLED_0.14.0-b3_ESP02_2306240.bin.gz](https://github.com/Aircoookie/WLED/releases/download/v0.14.0-b3/WLED_0.14.0-b3_ESP02_2306240.bin.gz)` to the release.
> > We still need to keep an eye on binary size, as the sum of b3 and b3 .gz combined is 1447, which is only about 50 kB below limit.
> 
> This compressed file (and WLED_0.14.0-b6_ESP02.bin.gz) are still not working for athom bulbs stuck on 0.13.0-b6.

I was able to flash 0.14.0-b6. But the latest one still fails for me. Is there anything we can do, or are we stuck with this version?

danielbrunt57, 2023-11-22:
> I was able to flash 0.14.0-b6. But the latest one still fails for me. Is there anything we can do, or are we stuck with this version?

FYI, I've abandoned the Athom WLED device as it is severely hampered by the 1M limit, plus I've had issued with WLED & Hyperion.  That and I found the Wi-Fi link between Hyperion & WLED a little slow.
I've replaced it with HyperHDR on a Pi 3B+ connected to a USB-C powered [Pimoroni Plasma 2040 (RP2040=based driver board](https://shop.pimoroni.com/products/plasma-2040?variant=39410354847827)) and my APA102 light strip for ambient TV lighting.  Check out if [HyperHDR](https://github.com/awawa-dev/HyperHDR) and [HyperSerialPico](https://github.com/awawa-dev/HyperSerialPico)  suits your needs.

danielbrunt57, 2023-11-22:
> Athom has an ESP32 version of the controller, you can choose that one, it has 8MB of flash memory

Thanks!
That would have been useful info a month or two ago but I am well underway now with my old Pi3B+ & HyperHDR (free) & Pico RP2040 (CAD$25) vs [Athom SKU: LS8P-WLED](https://www.athom.tech/blank-1/wled-esp32-music-addressable-led-strip-controller) (US$24.85).  Still, I'll keep it in the back of my mind.

Balooski, 2024-01-04:
Is anyone still having this issue? Last file I could update was 0.14.0-b6. Every update after that fails, even using the .gz provided.

danielbrunt57, 2024-01-04:
Not me.  I abandoned my WLED-based Athom LS-4P in favor of an RP2040-based Pimoroni Plasma 2040 directly connected to my Pi 3B+ running HyperHDR.

softhack007, 2024-01-04:
> Is anyone still having this issue? Last file I could update was 0.14.0-b6. Every update after that fails, even using the .gz provided.

Flash space for firmware is "on the edge" with ESP8266 2MB, and "over the edge" with ESP8266 <= 1MB.

If you have problems to OTA update your device with the .gz file, it could be that the device was pre-installed with a filesystem of >512Kb. In that case, you can only update by USB, but previously a complete chip erase with `esptool erase_flash` is necessary, removing the too-large filesystem. So make a backup of your old config and presets.

After complete erase, you must upload WLED by USB - OTA not possible, sorry.

igorsantos07, 2024-05-09:
I have two of these models, and gzipping 0.15.0-b2 from 900.5kb to 648kb did the trick :pray: 

Although I had to re-pair and restore backups, but I guess this is expected?