wled/WLED #3258: Off refresh doesn't work properly in build 2306210

Prowler1000, 2023-06-22:
### What happened?

I was notified of a new version release (v0.14.0-b3), and after updating, off refresh doesn't appear to work in some situations. The checkbox for it in LED Settings also doesn't seem to work.

### To Reproduce Bug

- Install [v0.14.0-b3](https://github.com/Aircoookie/WLED/releases/tag/v0.14.0-b3)
- Enable Off Refresh for an LED output (if possible, I am unable to disable it)
- Turn off all individual segments of said output.

### Expected Behavior

I am using TM1814s and normally, they would just be off. After the update, however, they flash as if they aren't receiving data if all individual segments of an output are turned off. 

If turned off with only the master (at least one segment is on), it operates as expected.

### Install Method

Binary from WLED.me

### What version of WLED?

2306210

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

boiler54, 2023-07-15:
I just installed 0.14.0-b3 and noticed the same behavior. When the main power button is pressed (turning off all segments), off refresh works correctly. But if you turn off an individual segment using the little power button on the segments page, off refresh does not work.

blazoncek, 2023-07-15:
I am not saying that you are not seeing incorrect behaviour, but the following code is used for off refresh:
```c++
    if (!offMode || strip.isOffRefreshRequired())
      strip.service();
```
There are no segments involved in off refresh handling.

Please provide oscilloscope or logic analyzer trace of the signal in such case.

blazoncek, 2023-09-27:
No feedback received.
Closing.

closed 2023-09-27 by blazoncek