wled/WLED #3259: SyntaxError when using "To the Controls" of the WLED-AP in first-boot setup

Craft4Cube, 2023-06-22:
### What happened?

When trying to use the "To the Controls" feature of the AP, the following error is displayed:

`SyntaxError: Unexpected token '<', "`

### To Reproduce Bug

Reset WLED to Factory Defaults. Connect to its WLED-AP. Click on "To the Controls"

### Expected Behavior

The Controls/Siliders page works as it should

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b3 (build 2306180)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

It seems WLED is trying to access /sliders/json/*** instead of /json/*** when using the first-boot screen

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-06-22:
Please open browser's developer tools and post network tab screenshot.
Possibly with incorrect response/request selected.
This may be due to recent reverse proxy support being added.

Craft4Cube, 2023-06-22:
[image]

blazoncek, 2023-06-22:
Thanks, I see the issue.

blazoncek, 2023-06-22:
@Craft4Cube can you check if adding the following line into index.js helps?
index.js @ line 220
```js
if (paths[0]==="sliders") paths.shift();
```

Craft4Cube, 2023-06-22:
Adding that line fixes it!

closed 2023-06-22 by blazoncek

klrock, 2023-07-14:
I have this exact same error using latest version 0.14.0-b3.  Has this error been corrected and is there a revision available to use?

akohli-86, 2023-07-15:
> Adding that line fixes it!

> Adding that line fixes it!

Can you help out? Tried putting in the line but seems it doesnt make any difference!

blazoncek, 2023-07-15:
As you can see from the commit above, the original issue has been resolved.
Please use latest nightly and clear browser cache and local storage.