wled/WLED #3261: Athom WLED Controller updated to b3 but now no wifi or way to reset/connect

swampylee, 2023-06-22:
### What happened?

I have the Athom WLED controller - https://www.athom.tech/blank-1/wled-esp32-music-addressable-led-strip-controller

I've been using a previous version of b2 without issue but this upgrade to b3 appears to have hung the Athom controller. I have orange LED's on the matrix but do not have any apparent Wifi - no access point found and didn't re-appear on my network as the previous mDNS entry.

If I disconnect the LED matrix and power back up it appears on the network, the settings look the same as before the update (Relay GPIO 2 Invert). I then unplug power and plug in the LED Matrix and then the controller hangs. Disconnect the LED's and the controller appears back on wifi. I've also tried the latest build with sound usermod and the same thing happens.

I've also factory reset, reinstalled the released B3, configured and the same happens again when the LED's are plugged in. So basically the controller is up and running as long as I don't plug the LED matrix in (16x16).

### To Reproduce Bug

Use a WLED Athom controller and upgrade to B3.

### Expected Behavior

As pre-B3 releases, once installed and the settings changed to Relay GPIO=2 and LED GPIO=18 it should work fine.

### Install Method

Binary from WLED.me

### What version of WLED?

both the released B3 and the latest B3 with sound usermod.

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

swampylee, 2023-06-22:
Went back to a b2 release (details below) and everything is working great, including the sound reactive usermod.

[image]

softhack007, 2023-06-22:
> So basically the controller is up and running as long as I don't plug the LED matrix in (16x16).

This part is unusual... if you have ws281x or similar, them WLED does not even check if LEDs are connected - data is written out "blindly". I often test with LEDs configured but none connected, as it makes no difference for WLED.

The only difference I can imagine is power consumption - could be that your LEDs are causing a voltage drop at power-on, and the esp32 detects "brownout" and reboots.

Please connect your device to a computer, and start serial monitor to check for messages from the controller.

swampylee, 2023-06-22:
Thanks for the response. I understand your thinking, it crossed mind too, but this exact hardware has been working fine and when I installed b2 back on there everything worked fine. I’d be happy to try and debug but the controller is back on b2 and working fine and I’m not sure how I would connect to a PC as it’s not a “normal” ESP32 dev board with a usb connector.

ndom91, 2023-06-22:
I'm having the same issue with an ESP32 and a 1M WS2812 running at 5V off the GPIO pins.

Previously running on `b1` for months without a problem. Now it does not seem to connect to wifi (or maybe even boot?) when the LED strip is attached. Plugged into that same power supply without the LED strip attached works fine, and plugged into the laptop works fine as well. 

I reinstalled the latest commit available today and the issue remains.

[image]

softhack007, 2023-06-24:
The "cannot use unless no LED connected" could be related to  this bug:
* https://github.com/Aircoookie/WLED/issues/3264

Fix is in work, should be availeable within the next week.

Scoff123, 2023-07-07:
Just came across this issue after updating from 13.3 - however I am no longer able to downgrade back to 13.3 - I am trying via the web installer, and selecting the correct ESP02 bin file, WLED_0.13.3_ESP02.bin, but I get an error saying Update Failed, and to check the file and retry. This happens no matter whether to try to flash WLED_0.13.3_ESP02.bin or WLED_0.14.0-b1_ESP02. 
I've factory reset my Athom controller and tried again after joining to my WiFi, but I still cannot downgrade - can anyone help please?

swampylee, 2023-07-07:
> Just came across this issue after updating from 13.3 - however I am no longer able to downgrade back to 13.3 - I am trying via the web installer, and selecting the correct ESP02 bin file, [...]

Also make sure you don’t have any LED’s attached so u plug power, disconnect LED’s and power back up and try updating.

Scoff123, 2023-07-07:
Thank you - I couldn't even get that version in the other issues solution to flash, so I've taken the board out of the enclosure and managed to successfully flash via serial and am now on version 0.14.0-b1 - however nothing works other than the web portal - no power function, no lights etc - just the web portal. The power button doesn't even do anything. I can change settings in the web portal and save them, which is strange. I may try flashing 0.13 and see if that restores normal functions

Scoff123, 2023-07-07:
It turned out I had to reset some of the GPIO mappings following the reflash that's all - including the GPIO for the relay. All working again now :)

swampylee, 2023-07-07:
> It turned out I had to reset some of the GPIO mappings following the reflash that's all - including the GPIO for the relay. All working again now :)

LOL, yes that’s caught me out before too!

Jamie-BitFlight, 2023-07-31:
I'm having the same symptoms on both the Quinled Quad v3.1 board and the ForkInEye pixelstick 3.

After updating to b3 release, it boots up, connects to WiFi, but no lights work, web interface doesn't load, iPhone app can't connect to it, ota update stop working.

Can only access it via usb connection for reflashing.

One of my forkineye pixelstick units which has a newer wemos board on it is responding after the update to b3, but the iPhone app interface is borked.
Screenshots below:

[image]
[image]

softhack007, 2023-07-31:
> the iPhone app interface is borked.

This is a well-known problem of the iOS app, not directly related to WLED.

> After updating to b3 release, it boots up, connects to WiFi, but no lights work, web interface doesn't load, iPhone app can't connect to it, ota update stop working.

Seems like a consequence of the brightness limiter issue on beta3 https://github.com/Aircoookie/WLED/issues/3264. Please wait until beta4 get released, or try a nightly build.

ovictorlorena, 2023-09-07:
> It turned out I had to reset some of the GPIO mappings following the reflash that's all - including the GPIO for the relay. All working again now :)

Can you do a step by step on this? It seems that my controller has this same problem, but I'm a little layman to know how to do this procedure

swampylee, 2023-09-08:
> > It turned out I had to reset some of the GPIO mappings following the reflash that's all - including the GPIO for the relay. All working again now :)
> 
> Can you do a step by step on this? It seems that my controller has this same problem, but I'm a little layman to know how to do this procedure

For Athom WLED - GPIO needs to be 18 for DATA1 and 17 for DATA2 – you also need to make sure that IR Info / GPIO Relay is set to 2 otherwise no lights.

ovictorlorena, 2023-09-08:
Thanks for your quick message. I had sent a message to the product manufacturer itself regarding this problem, they asked me to get these settings, so for the record that this is another possible solution when the controller is in perfect working order but does not emit light on the leds:
[image]

Scoff123, 2023-09-08:
Well done, you beat me to it - I got the screenshots and was annotating when I saw your reply!

softhack007, 2023-10-02:
Closing, as the original problem (too hight power draw due to broken auto brightness-limiter) was fixed with -b4.

Also I don't like how this problem report has degraded into a discussion of GPIO setting and other problems specific to Athom controllers. People, _please_ make an effort to stay on topic!

closed 2023-10-02 by softhack007