wled/WLED #3263: Bootloop when enabling usermod_v2_mode_sort

LorenKeagle, 2023-06-22:
### What happened?

When I attempt to build with this UserMod, my device gets stuck in a bootloop.

It seems that either this UserMod is not updated to reflect what appears to be some recent changes in the way lighting modes are stored, or there's a simple bug somewhere in the sort function from the stacktrace.  I don't believe my other usermods are related to this error, as it happens even when I comment those flags out.  

I'm not sure how bugs in usermods are handled, so I figured I'd let it be known here.

### To Reproduce Bug

Add "-D USERMOD_MODE_SORT" to your build flags.

### Expected Behavior

WLED should not crash, at the very least.  Expected functionality is that modes and palettes will be sorted by name.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0b3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4010dcd8  PS      : 0x00060b30  A0      : 0x800563e9  A1      : 0x3ffb1e50  
A2      : 0x00000000  A3      : 0x3ffb2540  A4      : 0x00000075  A5      : 0x00006880  
A6      : 0x000000ac  A7      : 0xffffffaa  A8      : 0x3ffb971c  A9      : 0x3ffb1e60  
A10     : 0x3ffb2528  A11     : 0x00001800  A12     : 0x00000001  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000019  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

ELF file SHA256: 0000000000000000

Backtrace: 0x4010dcd5:0x3ffb1e50 0x400563e6:0x3ffb1e70 0x400564c9:0x3ffb1e90 0x4010e5b5:0x3ffb1ee0 0x4010e61d:0x3ffb1f00 0x401b3c61:0x3ffb1f20 0x4011535d:0x3ffb1f40 0x4011619e:0x3ffb1f90 0x40135db2:0x3ffb1fb0 0x4008bcd2:0x3ffb1fd0
  #0  0x4010dcd5:0x3ffb1e50 in re_qstringCmp(void const*, void const*) at wled00/src/dependencies/json/ArduinoJson-v6.h:3945
  #1  0x400563e6:0x3ffb1e70 in ?? ??:0
  #2  0x400564c9:0x3ffb1e90 in ?? ??:0
  #3  0x4010e5b5:0x3ffb1ee0 in ModeSortUsermod::re_sortModes(char**, unsigned char*, int, int) at wled00/src/dependencies/json/ArduinoJson-v6.h:3945
      (inlined by) ModeSortUsermod::sortModesAndPalettes() at wled00/../usermods/usermod_v2_mode_sort/usermod_v2_mode_sort.h:149
  #4  0x4010e61d:0x3ffb1f00 in ModeSortUsermod::setup() at wled00/src/dependencies/json/ArduinoJson-v6.h:3945
  #5  0x401b3c61:0x3ffb1f20 in UsermodManager::setup() at wled00/um_manager.cpp:7 (discriminator 3)
  #6  0x4011535d:0x3ffb1f40 in WLED::setup() at wled00/wled.cpp:408
  #7  0x4011619e:0x3ffb1f90 in setup() at wled00/wled00.ino:16
  #8  0x40135db2:0x3ffb1fb0 in loopTask(void*) at /Users/lkeagle/.platformio/packages/framework-arduinoespressif32@src-e9b1fbd6563a55e19ddae15e1fc09589/cores/esp32/main.cpp:18
  #9  0x4008bcd2:0x3ffb1fd0 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jul 29 2019 12:21:46

[... 104 lines cut ...]

Rebooting...
ets Jul 29 2019 12:21:46

etc...
```

### Anything else?

The board is a Laterna controller that I'm trying to augment with 4 line display and PIR sensor.  

Here are all my current flags in my override file:

build_flags = ${common.build_flags_esp32} 
  -D WLED_RELEASE_NAME=ESP32 
  -D WLED_DISABLE_BLYNK 
  -D USERMOD_PIRSWITCH
  -D USERMOD_MODE_SORT
  -D USERMOD_FOUR_LINE_DISPLAY -D USE_ALT_DISPlAY -D FLD_PIN_SCL=33 -D FLD_PIN_SDA=32
  -D USERMOD_AUTO_SAVE -D AUTOSAVE_PRESET_NUM=1
 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-06-23:
Please contact original usermod author to fix it.

On a side note, Sort usermod is no longer needed for ALT usermods as is setting GPIO pins using FLD_PIN_SCL and FLD_PIN_SDA.
For I2C interface there are now global variables for it and global constants I2CSDAPIN and I2CSCLPIN.

LorenKeagle, 2023-07-13:
I never knew it was needed originally for the ALT usermods.  I simply wanted to have the modes on my 4 line display sorted alphabetically, in order to navigate to known effects more easily.  

I'll disable for now, but I'll figure out how to tag the original author when I get a chance.

LorenKeagle, 2023-07-13:
@blazoncek It seems you're the original author of this usermod.  It seems to no longer serve its original purpose, however it would be nice to still be able to see alphabetically sorted modes in a display.

At the very least, if it truly is no longer needed, updating the documentation and cleaning up the example build flags to not indicate that it is needed would be helpful for those of us just starting out on custom WLED builds.

blazoncek, 2023-07-13:
I am not an original author of these usermods, though I am maintaining ALT usermods ATM.

ALT usermods are an evolution of original rotary enccoder and 4 line display.

As for documentation, you can help update it after you resolve your issues. The instructions for successful compilation should be included in `readme.md` accompanying the usermod source.

As noted above, you no longer need sort usermod if you use ALT usermods.

Edmundajw, 2023-09-04:
Hi @blazoncek, I've just come across this thread after struggling for the last few hrs with rotary_encoder_ui, which has been causing a boot loop trying to use it without the alt funcitonality- Not specifiying -D USE_ALT_DISPLAY, and specifying MODE_SORT.

I understand that _ALT is the updated, latest version, and I have that working fine. I want to use the original, as, using no display, I just want to use the encoder switch to cycle a few modes, and L/R to go through them. Having no display, it's very hard to properly use the alt version- and I want a simple/dumb solution.

I could probably spend some time and modify the alt version to work on clicks for mode changes, as per the first version, but it seemed to do exactly as I needed off the bat..so was hoping I could just use it as is..

Any idea on a simple solution? I realise it's probably not THAT hard to code up- I'm just new to the language... spent weeks trying to build my own stuff off NeoPixel:Matrix in Arduino IDE, and just found this project- so seeing what I can cobble together with existing mods...before I accidentally try to reinvent the wheel again...

Cheers,
Edd

blazoncek, 2023-09-04:
Hi @Edmundajw, as noted earlier non-ALT version is unmaintained by core WLED team. You will need to talk to original author of the usermod to bring it up to speed or try yourself with the task (and when you succeed perhaps make a PR back upstream).

0.14 had a fundamental change how effect names are stored and hence made sort usermod inoperable. As the ALT version also relied on it I decided against updating sort usermod but instead moved sort functionality directly into ALT version. This decision was made due to the limited number of usermods that are allowed to run simultaneously and sort was using one of those slots without providing benefit for any other usermod.

You may try to extract sorting from ALT version and move it into non-ALT and remove the dependency from sort usermod entirely.

And BTW ALT does everything non-ALT does. And more.

closed 2024-01-02 by blazoncek