wled/WLED #3265: Individual segment turn off turns them dim rather than off, after update to 0.14.0-b3 on ESP32

ingestre, 2023-06-23:
### What happened?

After update to 0.14.0-b3, when I issue the command to turn segments off - the LED's dont turn off completely, but instead go very dim. If I run the command a second time they turn off fully.

This seems to happen if I use the Toggle segment command as well.

Downgrading back to 0.14.0-b1 fixes this issue. 

### To Reproduce Bug

API Command I use to turn all segments on as follows

`{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":49,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,183,115,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":1,"start":49,"stop":176,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,183,115,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"id":2,"start":176,"stop":244,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,183,115,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

API Command I use to turn all segments off as follows

`{"seg":[{"id":0,"start":0,"stop":49,"grp":1,"spc":0,"of":0,"on":false,"frz":false},{"id":1,"start":49,"stop":176,"grp":1,"spc":0,"of":0,"on":false,"frz":false},{"id":2,"start":176,"stop":244,"grp":1,"spc":0,"of":0,"on":false,"frz":false}]}`

API Command I use to toggle a segment as follows

`{"seg":[{"id":2,"on":"t","col":[[255,183,115,0],[0,0,0,0],[0,0,0,0]]}]}`

### Expected Behavior

Running the toggle command turns the third segment dim and on rather than off and on

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b3 (build 2306210)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

GUI peek seems to show the LED's as off when they are dim.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-06-23:
Please try 2 different settings: transition set to 0 and transition set to 1s and report back.

ingestre, 2023-06-23:
With transition set to 0 the toggle command and the off command seems to do nothing. LED's remain lit.

With transition set to 1s (10) the off command turns the LED's dim, and the toggle command toggles betweeen on and dim.

Hope this helps.

blazoncek, 2023-06-23:
One more question: Are you seeing this behaviour only with Solid effect?

ingestre, 2023-06-23:
No - Other effects seem to exhibit it also.

I tried it with the following setting which sets "Fire 2012" (fx66) on segments 1 & 3

`{"on":true,"bri":128,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":49,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[0,0,0,0],[0,0,0,0],[0,0,0,0]],"fx":66,"sx":128,"ix":128,"pal":35,"sel":false,"rev":false,"mi":false},{"id":1,"start":49,"stop":176,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,183,115,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":35,"sel":true,"rev":true,"mi":true},{"id":2,"start":176,"stop":244,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,160,0,0],[0,0,0,0],[0,0,0,0]],"fx":66,"sx":128,"ix":128,"pal":0,"sel":false,"rev":true,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

I then used the following API code to toggle segment 3

`{"seg":[{"id":2,"on":"t"}]}`

When I toggle the segment off the animation halts on that segment, but the LED's that are lit go dim rather than off. (The animation normally only lights one side of the segment to simulating fire coming from that side)

When I toggle again the animation starts and the LED's go back to usual brightness

arunankam, 2023-06-26:
Same issue... LED's dim instead of turning off.

blazoncek, 2023-06-26:
@arunankam as you can see from above this is fixed in the source. Will be available in next release.

ingestre, 2023-08-13:
Just installed the latest release [v0.14.0-b4 "Hoshi"] and I can confirm that upon initial inspection the bug appears fixed.

Thank you blazoncek for your prompt attention, and all the developers for your hard work.

blazoncek, 2023-08-13:
Please close then.

ingestre, 2023-08-13:
Closed as fixed.

closed 2023-08-13 by ingestre