wled/WLED #3268: Leds real time color update

RenautMestdagh, 2023-06-24:
**Is your feature request related to a problem? Please describe.**
I don't really like it the  color of the leds only change when you release your finger from the color picker.

**Describe the solution you'd like**
I'd like it if the leds change color real time while still dragging your finger on the color picker.

**Describe alternatives you've considered**
Maybe a toggle in the options to enable/disable this behaviour?

**Additional context**
/

Thank you for your ideas for making WLED better!

blazoncek, 2023-06-24:
This is not implemented (and will not be) because MCU is not designed to handle flood of HTTP/WS requests which would be caused by such behaviour.
At least until more powerful ESP comes along with plenty of RAM.

moon4-hacs, 2023-06-24:
But if you control WLED via Home Assistant it works...

blazoncek, 2023-06-24:
> But if you control WLED via Home Assistant it works...

You are being lucky.

closed 2025-01-27 by RenautMestdagh