wled/WLED #3269: Request rf remots 433mh

th3alex14, 2023-06-24:
there are some 433mh led remote controls. Is there a way to use them.
so you can have longer range

blazoncek, 2023-06-24:
Not without additional hardware and custom usermod.

th3alex14, 2023-06-25:
i got the hardware and the codes in binary or hex of the remote and tried to write a user mod maybe someone with programming knowledge could take a look 

#include <RCSwitch.h>
#include "wled.h" 
#include "led.cpp"
#include "ir.cpp"
#include "ir_codes.h"

 RCSwitch mySwitch = RCSwitch();
 #define on_1  2803457
 #define auto_1  2803458
 #define Off_1  2803459
 #define S_plus  2803460
 #define M_plus  2803461
 #define brig_plus  2803462
 #define S_minus  2803463
 #define M_minus  2803464
 #define brig_minus  2803465
 #define red  2803466
 #define green  2803467
 #define lila 2803468
 #define yellow 2803469
 #define yellow2 2803470
 #define orange 2803471
 #define baech 2803472
 #define blue 2803473
 #define blue2 2803474
 #define lila2 2803475
 #define green2 280376
 #define white 2803477
 #define Pin 23

class rfresiver : public Usermod {
private:
  uint8_t lastValidCode = 0;

public:
  void setup() {
    Serial.begin(9600);
    mySwitch.enableReceive(32);
  }

  void loop() {
    if (mySwitch.available()) {
      uint32_t value = mySwitch.getReceivedValue();
      if (value != lastValidCode) {
        Serial.print("Received value: ");
        Serial.println(value);
        lastValidCode = value;

        switch (value) {
          case on_1:
            bri = briLast;
            break;
          case Off_1:
            toggleOnOff() ; 
            break;
          case auto_1:
            presetFallback(3, FX_MODE_FIRE_FLICKER, 0); 
            break;
          case S_plus:
            changeEffectSpeed( 16);
            break;
          case S_minus:
             changeEffectSpeed(-16); 
            break;
          case M_plus:
            changeEffect(relativeChange(effectCurrent, 1, 0, strip.getModeCount() -1));
            break;
          case M_minus:
             changePalette(relativeChange(effectPalette, 1, 0, strip.getPaletteCount() -1));
            break;
          case brig_plus:
            incBrightness(); 
            break;
          case brig_minus:
            decBrightness();    
            break;
          case red:
            changeColor(COLOR_RED);   
            break;
          case green:
             changeColor(COLOR_GREEN);  
            break;
          case lila:
            changeColor(COLOR_PURPLE); 
            break;
          case yellow:
            changeColor(COLOR_YELLOW);
            break;
          case yellow2:
            changeColor(COLOR_YELLOWISH);
            break;
          case orange:
            changeColor(COLOR_ORANGE); 
            break;
          case baech:
            changeColor(COLOR_WARMWHITE2); 
            break;
          case blue:
            changeColor(COLOR_BLUE); 
            break;
          case blue2:
            changeColor(COLOR_DEEPBLUE); 
            break;
          case lila2:
            changeColor(COLOR_PINK);   
            break;
          case green2:
            changeColor(COLOR_GREENISH); 
            break;
          case white:
            changeColor(COLOR_WHITE);  
            break;
          default:
            break;
        }
      }
      mySwitch.resetAvailable();
    }
  }
};

blazoncek, 2023-06-25:
Take a look at #3237

dosipod, 2023-06-28:
> Take a look at #3237
Is it possible to connect two MCUs running wled  with espnow

blazoncek, 2023-06-28:
Not yet, but perhaps in the future.

th3alex14, 2023-06-28:
I'm trying to write a usermod but for some reason the rc-switch libery works on esp32-3 but not on esp32 wroom 
Many i figure it out

softhack007, 2023-07-15:
You can grab a usermod here:
https://github.com/athom-tech/Sound-Reactive-WLED/tree/master/usermods/RF433