wled/WLED #3273: ESP8266 Random Resets & Weird Relay Behavior at Boot

kooopa, 2023-06-28:
### What happened?

First, I thank Everyone involved in this project for making this. It's freaking awesome.

Context:
I built a control board running Wled to control Addressable LEDs to be installed on my motorcycle. I used 3 different Buck converters to power the LEDs and the ESP8266.

Issue #1 ESP8266 resets.
Behavior: Controller resets when rapidly clicking buttons or other LED control buttons on the main page.

Note: A V1 usermod was added to change presets using a combination of GPIO 12,13 states for a total of 4 presets. Rapidly cycling through the presets using this external input doesn't trigger a reset.

Note: Issue not present on 0.14.0-b1 release
___________________________________________________________________________________________________________________________________________
Issue #2 Relay doesn't switch properly when "Turn LEDs on after power up/reset" is unchecked, and then the relay is switched on using the boot preset. Relay Pin: GPIO 5

Behavior: Relay Switches On then Off rapidly at boot, then stays off despite the power indicator in the web UI indicating that it's on. Selecting another preset or Toggling the Power Button on the web UI triggers the relay properly.

The relay is used to remove the momentary flicker when the ESP is booting up, and there is garbage on the data lines.

Note: I have yet to try on the 0.14.0-b1 release.

### To Reproduce Bug

Issue#1: Rapidly change settings on the main page: i.e., presets, segments, and colors. Etc.

Issue#2: Set Turn LEDs on after power up/reset unchecked and turn relay/LED's on using a preset to be triggered at boot.

### Expected Behavior

Issue#1
Expected Behavior: WLED shouldn't reset while rapidly changing presets or led states.

Issue#2
Expected Behavior: The relay should be off at boot and turn on after the set boot preset is played(w/ initial 1sec delay using a playlist and a preset).

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b3 (2306270)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-06-28:
There's nothing that can be done for #1 as the 8266 MCU is incapable of handling quick UI requests. Switching to ESP32 may help.
I cannot reproduce #2 on any of my test set-ups. I am using professionally built WLED controllers from @srg74 , @intermittech (AKA Quindor) and @wled-install as well as home built test rigs.
Note: I primarily use GPIO12 for relay but GPIO4, GPIO5 and GPIO13 are also working as expected. Make sure, you pull the relay GPIO low.

kooopa, 2023-06-28:
As for #1, I compiled release WLED 0.14.0-b1, and it's not present there. No matter how fast I click, it doesn't reset the controller. And it seems pretty stable too. On 0.14.0-b3 it randomly throws JSON error. I'm still testing it, but no issues so far.

As for #2, it is pulled low using a 10K resistor. I looked at @srg74 dev board circuit, and coincidentally I'm using the same exact circuit and components for the relay driver. I'll switch the GPIO pins for the relay and see if it makes a difference.

kooopa, 2023-06-28:
Update on #1: I managed to break 0.14.0-b1 by changing presets and settings rapidly, but it's very hard. You must click fast like you're intentionally trying to break it, but the controller doesn't reset. Web UI just freezes and usually recovers after a second or two.

Basti149, 2023-09-06:
Hello, I have the same problem as "kooopa" with the WLED version 0.14.0-b4 "Hoshi" on the ESP12. It also reboots suddenly when quickly pressing the on and off button on the GUI or changing the color/effect or settings.

Like you, I have the second error, namely the relay lights up/switches on or the LED lights up. Despite the fact that he is out in the gui. I use the LED on pin D4.

If you do nothing on the controller/ ESP12/ Wemos and the WS2812 LEDs are off. Then the ESP works without any problems and no restarts are carried out.

Is there a solution to the problem in the meantime?

x additional capacitor
x older Wled version or would it be possible to use another software instead of WLED.

I've tested it with different ESPs and boards, with the controller H801, ESP12, ESP07s, Wemos, LuanBoard, the same problem everywhere.

I hope there is a solution maybe you can remove a few effects eg 2D from the software so that it works again because it is impossible and useless.

Hallo das gleiche Problem wie "kooopa" habe ich mit der WLED Version 0.14.0-b4 „Hoshi“ auf dem ESP12 auch. Er startet beim schnellen Drücken des Ein und Aus Button auf der GUI oder bei Änderung der Farbe/ des Effekts oder der Einstellungen bootet er auch plötzlich neu.

Den zweiten Fehler habe ich auch wie du und zwar Leuchtet/ Schaltet das  Relais Ein bzw leuchtet die LED. Trotz das er in der Gui aus ist.  Ich nutze die LED an den Pin D4.

Wenn man an den Controller/ ESP12/ Wemos nichts macht und die WS2812 LEDs aus sind. Dann funktoniert der ESP ohne Probleme und es werden auch keine neustarts ausgeführt. 

Gibt es in zwischen eine Lösung zum Problem? 

x zusätzlicher Kondensator
x ältere Wled Version oder wäre es möglich auch eine andere Software zu nutzen anstatt WLED.

Ich habe es mit verschiedenen ESPs und Platinen getestest und zwar mit dem Controller H801, ESP12, ESP07s, Wemos, LuanBoard überall das gleiche Problem.

Ich hoffe es gibt eine Lösung vielleicht kann man ein paar Effekte zb 2D aus der Software entfernen damit sie wieder funktoniert den so ist es unmöglich und unbrauchbar.

blazoncek, 2023-09-06:
What is ESP12? If you are referring to ESP8266-12E (or F) then it is incapable of responding to quick HTTP requests and will (usually) reset. As a remedy try switching to ESP32.

For the second verify boot states of the GPIO used for relay. Some GPIO will output gibberish at boot and some require to be pulled-up - hence relay switch during boot. Use another GPIO in such case or try adding strong pull-down (but I do not think this will help).

None of the above can be solved in software.

Basti149, 2023-09-06:
> What is ESP12? If you are referring to ESP8266-12E (or F) then it is incapable of responding to quick HTTP requests and will (usually) reset. As a remedy try switching to ESP32.
>
> For the second [...]

Yes, I meant the ESP, this is colloquially just called ESP12 (see espressif/inscription on the ESP).
Since the LEDs should only light up, in my opinion an ESP12 is sufficient instead of a 32.

With version 0.13.3, Wled still works without errors, only from higher there are such problems. Since the ESP12/8266 works with Tasmota without any errors, I am 100 percent sure that the error is in the WLED version. Maybe it would help to remove some useless codes that nobody needs. For decent lighting just 2D integration or music can be removed.
 
As an output, I only use the integrated LED on D4, which I switch on/off. Which unfortunately no longer works after a while. And since I tested it on different ports and other users have this problem too. As I said, it is due to the programming of the WLED code, maybe it has become too messy or too complex.

Ja den ESP meinte ich dieser wird umgangsprachlich nur ESP12 genannt (siehe Espressif/aufschrift auf dem ESP).
Da die LEDs nur leuchten sollten reicht meiner Meinung nach einen ESP12 aus anstatt eines 32.

Unter der Version 0.13.3 funktoniert Wled noch ohne Fehler erst ab höher gibt es dermaßen Probleme. Da der ESP12/8266 bei Tasmota ohne Fehler funktoniert bin ich mir 100 prozent sicher das der Fehler in der WLED Version liegt. Villeicht würde es helfen ein paar unnützige Codes zu entfernen die keiner braucht. Für eine anständige Beleuchtung gerade die integration 2D oder Musik kann man entfernen.
 
Als Ausgang nutze ich nur die integrierte LED an D4 die ich Ein/ Ausschalte. Die leider nach einer Weile nicht mehr funktoniert. Und da ich das auf verschiednen Ports getestet habe und auch andere User dieses Problem haben. Liegt es wie gesagt an  der Programmierung des WLED Codes vileicht ist er zu unordentlich oder zu komplex geworden.

softhack007, 2023-09-06:
> Ja den ESP meinte ich dieser wird umgangsprachlich nur ESP12 genannt

Hello @Basti149 please use english language when replying. @blazoncek is not german, and I find it quite unfair that you expect him to do the translation.

Ich bin im moment der einzige im Team der Deutsch spricht. Es bringt also nicht auf Deutsch zu antworten.

Right now I'm the only german in  the dev team, so it won't help to answer in german.

softhack007, 2023-09-06:
> Maybe it would help to remove some useless codes that nobody needs.

Iwhat do you mean here ? f you have identified unnecessary and unused code, please tell us exactly which lines could be deleted. Preferably with pull request that can be reviewed by the team.

Otherwise -  if this this was just a "figure of speech" - please avoid that in future. It does not help, and it just makes everyone wonder what are you talking about.

softhack007, 2023-09-06:
> managed to break 0.14.0-b1 by changing presets and settings rapidly, but it's very hard.

This is a known issue in beta  versions earlier than -b4. Please try again with 0.14.0-b4.

blazoncek, 2023-09-06:
@Basti149 there is DMX I do not use, also ArtNet and E1.31. Hue and Alexa give me vomits and I could go on, but there are plenty of users demanding each of those.

Each and every line in WLED is there for a reason and a lot of obsolete code has been pruned already.

MattParkerDev, 2023-09-06:
Hi, I’m getting the same issue on my ESP8266. Is it possible to rate limit the requests a client sends to the ESP8266 given that this is an inherent limitation of this specific controller? 

My repro is tapping around on the colour circle trying to find a colour, and the board would reset.

blazoncek, 2023-09-07:
> Is it possible to rate limit the requests a client sends

Yes, but that is *client's* responsibility. Unfortunately the library WLED uses for HTTP requests tries to process everything (and it will do so until RAM is depleted).

Basti149, 2023-09-08:
> > Ja den ESP meinte ich dieser wird umgangsprachlich nur ESP12 genannt
> 
> Hello @Basti149 please use english language when replying. @blazoncek is not german, and I find it quite confusing that you expect him to do the translation, or add the same sentences in German which is no added value.
> 
> Ich bin im moment der einzige im Team der Deutsch spricht. Es bringt also nichts auf Deutsch zu antworten.
> 
> Right now I'm the only german in the dev team, so it won't help to answer in german.

I posted it in both English and German. There's something for everyone :)

Basti149, 2023-09-08:
> What do you mean here ? If you have identified unnecessary and unused code, please tell us exactly which lines could be deleted. Preferably with pull request that can be reviewed by the team.
> 
> Otherwise - if this this was just a "figure of speech" - please avoid that in future. It does not help, and it just makes everyone wonder what are you talking about.
> 
> > As I said, it is due to the programming of the WLED code, maybe it has become too messy or too complex.
> 
> Another one. Please avoid such comments, it only makes you look like a fool. Which I'm sure is not what you want to be.

I meant that the function of the WLED.bin of the ESP8266/ ESP8265/ ESP 12 is too complex, for example the 2D, microphone integration would be a thing that you don't necessarily need with the ESP variant. If you omit this function in the ESP8266's FW. Then the module will definitely run more smoothly and won't crash as quickly as in my case and that of other users. A limit on HTTP requests would also be possible. I'm looking for a way to use this WiFi module for simple WLED things.

For highly complex tasks there is the ESP32 that can do it all. Which I also have running with the current one without any problems.
But in some projects an ESP32 is too oversized for a standalone color or as a running light.

The whole thing should be a suggestion not a criticism because it doesn't just affect me.

Basti149, 2023-09-08:
> @Basti149Es gibt DMX, das ich nicht verwende, außerdem ArtNet und E1.31. Hue und Alexa geben mir Erbrochenes und ich könnte weitermachen, aber es gibt viele Benutzer, die jedes dieser Dinge verlangen.
> 
> Jede einzelne Zeile in WLED hat einen bestimmten Grund und eine Menge veralteter Code wurde bereits bereinigt.
Alexa, Hue integration is good. But what if you removed the 2D or 3D effects from the ESP8266? Does it then work without errors? I'm currently on 0.13.3 and there are no crashes.
I could imagine that something new (2D) has been added that the ESP module doesn't like.

I'm just talking about the ESP12/8266 etc.

With the Esp 32 everything works very well and great.

blazoncek, 2023-09-08:
You can always compile from source and disable certain components (including 2D).
Public binaries will include most common features majority wants/needs. If you are not satisfied with those, learn how to compile source code and do so.

I am closing this issue as rapid clicking to bring ESP to its knees is beyond our control.
If you think this *has* to be solved then, please, open an issue with AsyncWebServer.

closed 2023-09-08 by blazoncek

Basti149, 2023-09-08:
> You can always compile from source and disable certain components (including 2D). Public binaries will include most common features majority wants/needs. If you are not satisfied with those, learn [...]

I don't agree with the closure. Since the problem doesn't just affect me, the precompiled bin for the 8266 on Github in its current version would be pointless and useless. Please check suggestion. Since the error occurred from the last version, I assume this is the case. That a wannabe layman messed something up. Maybe he should have tested his own project first instead of installing it in WLED.

softhack007, 2023-09-08:
> That a wannabe layman messed something up. Maybe he should have tested his own project first instead of installing it in WLED.

Talking about yourself ? Because I know many contributors of WLED, and none of them is a "wannabe layman". Cool down, calm your t***, man. _Es reicht_. Enough of this.

Standard 8266 binaries are as they are. If you want other WLED "variants" with features switched off, take a look at firmware repositories like this one: https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware, or compile WLED sourcecode yourself in VSCode+Platformio. There are many switches to remove features like 2D, Alexa, infrared, esp-now, etc etc.

I agree with @blazoncek this topic stays closed. 

About your other suggesions:
* you can compile WLED yourself to get a firmware without 2D support
* microphone support is not possible on 8266
*the features added since 0.13.3 do definitely not contain  something that "8266 does not like". As @blazoncek explained, we have some problems due to limited RAM on 8266, and the  AsyncWebserver library does not handle high rates of requests well. It is like it is.