wled/WLED #3276: Android Effect frame rate limit / display/encoder usermod interface

Proto-molecule, 2023-07-01:
### What happened?

The Android effect seems to not be affected by the frame rate limiter. Its missing "return FRAMETIME;" at the end of the effect. Im not sure of this was omitted on purpose. This doesn't seem to be a problem until the LED strip starts approaching 500 in length. At that point, as soon at you select the "Android" effect, the usermods become unresponsive.

### To Reproduce Bug

Compile with "usermod_v2_four_line_display_ALT" and "usermod_v2_rotary_encoder_ui_ALT" and enable
set the number of LEDs to 500+
select the Android effect using the web ui or the encoder
at this point the 4line display will freeze and the encoder will be unresponsive
select a different effect using the web ui
now the usermods will be functional again

### Expected Behavior

at this point the 4line display will freeze and the encoder will be unresponsive
select a different effect using the web ui
now the usermods will be functional again

### Install Method

Self-Compiled

### What version of WLED?

Latest 13 or 14

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-07-02:
The return value is as expected for the effect. Not all effects will return FRAMETIME as some depend on the exact amount of time to pass between calls. Android is such effect.

I am unable to reproduce the behaviour of 4LD and rotary encoder with two outputs of 512 pixels each. I do see a reduced frame rate of 15 though, using current development branch (with fixes for ABL).

Proto-molecule, 2023-07-02:
Hmm that’s interesting, mine reproduces even with two outputs of 270. I’ll do some more poking around.
If I set the framerate too high for everything then all the effects freeze the usermods.

softhack007, 2023-07-02:
> Hmm that’s interesting, mine reproduces even with two outputs of 270. I’ll do some more poking around. If I set the framerate too high for everything then all the effects freeze the usermods.

Please pull my latest commit and try again.
The encoder needs to be polled very often (each 8ms or more), so if could happen that frequent LED updates were preventing to read the encoder pins in due time.

Proto-molecule, 2023-07-02:
Thanks. I actually implemented a similar fix for myself. However to be completely honest I was missing a few commits in my tests. [blazoncek](https://github.com/blazoncek) when you mentioned you weren't having problems, that's when I noticed.  Android wasn't the only one causing problems. Thank you for the help and sorry for the trouble!

closed 2023-07-02 by Proto-molecule