wled/WLED #3277: Add support for FW1906 (introduced in NeoPixelBus 2.7.6)

Robert-github-com, 2023-07-07:
There is a new programmable PWM led driver FW1906 that support six LEDs per pixel. It is used in RGB+CCT LED strips. It would be great to add support for it to WLED.

Support for that chip was added recently to NeoPixelBus. So it requires to bump version to [2.7.6](https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.6) and use new features RgbwwColor & GrbcwxFeature.

There are some more information [here](https://wled.discourse.group/t/add-support-for-fw1906-not-the-same-as-tm1906-enhancement/8531).

blazoncek, 2023-07-07:
See #3256 about NPB 2.7.6

homeusepb, 2023-07-26:
Any updates on this?

blazoncek, 2023-07-26:
Please follow PR #3298 for updates.

Robert-github-com, 2023-07-26:
Hi,
No progress so far. I have a "reverseBalanceFromKelvin" function, but for
precise calculations 16bit white RGB color is needed. I tested some values
using 8bit RGB and the error is up to 3 per channel. But still better than
without.

blazoncek, 2024-03-24:
available since 0.15.0-b2

closed 2024-03-24 by blazoncek