wled/WLED #3279: Support for Home Assistant Bluetooth Proxy?

incaqueen, 2023-07-07:
Any thought about adding Home Assistant Bluetooth Proxy to WLED? I have a few WLED controllers and having this support would avoid me having to add a second ESP32 with ESPHome to that location.

It would be similar to the Bluetooth Proxy support that Shelly provides:
https://www.home-assistant.io/blog/2022/12/07/release-202212/#bluetooth-proxy-using-shelly-devices

blazoncek, 2023-07-07:
Bluetooth, if implemented, would cause image size to grow beyond capacity of 4MB devices.

yousaf465, 2023-07-20:
esphome is using ESP-IDF to conserve space.

scoobydoofus, 2024-01-24:
Just adding my vote. I would also love this feature if it was possible! I have a few BT Proxies around the place now, some in the same room as WLED devices.

kellerbaum, 2024-02-08:
Upvoting this as well. Would be useful as an option for devices with more than 4MB of storage.

DarkAssassinUA, 2024-04-09:
+vote

jeells102, 2024-07-26:
+vote - this will be useful for me as I have a ESP32 board in my living room and would be good to just have a Bluetooth proxy running alongside WLED.

mediocaballero, 2024-10-03:
Upvoting! This would be great to sace duplicate decices in some rooms!

softhack007, 2024-10-04:
I think this will not be possible on esp32.
* we are already "on the edge" of program space, so we cannot add any Bluetooth features for the normal 4MB devices
* the bluetooth proxy from EspHome needs a lot of RAM, and  eats lots of cpu time. I think it's not possible to have it running on the same board as WLED.

mediocaballero, 2024-10-04:
> I think this will not be possible on esp32.
>
> * we are already "on the edge" of program space, so we cannot add any Bluetooth features for the normal 4MB devices
> * the bluetooth proxy from EspHome [...]

oooh, it’s a pity…
anyway, thanks for dropping by and for the feedback! 👍🏻

netmindz, 2024-10-04:
Microcontrollers should really be seen as single-purpose devices. Being a Bluetooth proxy for another platform had nothing to do with WLED

yousaf465, 2024-10-05:
> I think this will not be possible on esp32.
>
> * we are already "on the edge" of program space, so we cannot add any Bluetooth features for the normal 4MB devices
> * the bluetooth proxy from EspHome [...]

we could make this an optional feature for devices with large RAM and space, such as ESP S3 devices being manufactured.

netmindz, 2024-10-05:
As mentioned it's not just about ram but CPU time as well

When an ESP32 is now less than 5 USD then there is no point trying cram multiple unrelated functions into a single device, which is then likely to perform badly in both roles

closed 2024-10-05 by netmindz