wled/WLED #3281: RGB analog strip, colors light up with different brightness

philxws692, 2023-07-09:
### What happened?

When I try the red color, the strip glows very dark. With the blue color it gets a little bit brighter. The green color is the most brightest (I don't know if it is the maximum but it's quite bright).
I know that the LEDs are not defect, as they light up the same brightness when I connect them to the Ground directly.

### To Reproduce Bug

I connected my analog RGB LED Strip according to [this](https://kno.wled.ge/basics/getting-started/) instruction. I added the 10k Ohm resistors and connected every color the same way. Then I just tried every color Red, Green and Blue without any mixing, just those three colors.

### Expected Behavior

I expect that all colors got the same brightness.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

If I forgot a relevant information please let me know :)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-07-09:
Switch GPIO and compare if the same brighness drop is seen.
Or make 3 PWM white channels and set all 3 to the same level and then compare.

blazoncek, 2023-09-18:
@philxws692 it would be nice if you could provide feedback.

blazoncek, 2023-09-27:
No feedback provided.
Closing.

closed 2023-09-27 by blazoncek