wled/WLED #3282: Crossfade stuttering with Solid Effect

Simonh98, 2023-07-10:
### What happened?

When using the solid effect and the 2D configuration, I noticed that the crossfade does not run cleanly and hangs when turned off or on. This behavior does not occur with any of the other (2D) effects.
However, I am not sure if the 2D configuration causes this behavior. It was not observed before the last two versions.

### To Reproduce Bug

Using the solid effect, possibly in combination with the 2D configuration

### Expected Behavior

Smooth crossfade

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b3 (2306180)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

[This video](https://youtube.com/shorts/MooZ6HinfzE) shows the described behaviour. The 2D aurora effect (green) is used first, followed by the solid effect (orange). It can be seen that the solid effect is dimmed to the lowest level, held for a short time and then abruptly switched off. Sometimes there is also a stuttering transition (not visible in the video).

LED Preferences:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-07-10:
Solid effect updates LEDs only 3 times per second while other effects usually do it about 42 times per second.
When you make a change to segment (color, brightness, effect, etc) the segment will enter a transition and will try to speed up effects but that may fail and regular speed is maintained.

softhack007, 2023-09-06:
@Simonh98 the upcoming "-b5" release comes with a major update of the "transitions" (cross-fade) feature.
Please check if the problem still exists when using the latest nightly build, and report back.
Thanks.

blazoncek, 2023-09-18:
@Simonh98 please provide feedback.

Simonh98, 2023-09-19:
> @Simonh98 please provide feedback.

Looks good. Couldn't reproduce the old behavior in the new version yet. Seems to work, ty!

closed 2023-09-19 by Simonh98