wled/WLED #3285: Adjustable NTP polling interval

rodhammond, 2023-07-10:
Sorry if I missed a config but assume NTP is used to set the internal HW time IRQ. I've noticed my animations briefly pause/freeze when NTP is enabled (found tip to disable NTP from QuinLED). Is it possible to allow the user provides a recommended NTP background polling interval in the form of hh:mm? I assume, after device boots, a fixed number of NTP requests will set time, then the user's hh:mm interval would kink in to maintain time. This would prevent the need to disable NTP to prevent animations from freezing/pausing. Please let me know if there is a different solution. Thanks

Aircoookie, 2023-07-11:
Indeed the NTP interval (once the time is successfully optained once) is hard-coded to 11 hours and 40 minutes. I don't see how having this interval user-configurable would prevent the brief freezes. Perhaps it would be more useful to have NTP poll only when the LEDs are off to prevent freezes, unless the LEDs are on for an extended amount of time (> 24 hrs ?). As I would assume most setups not to be turned on 24/7 this would eliminate random freezing due to NTP in most practical scenarios.

On a related note, the retry interval if there is no network access or the NTP server is unreachable is quite aggressive, leading to frequent 7 second freezes if there is a connection problem. This would be a separate issue to address, with potential mitigations being longer retry intervals or even temporarily disabling NTP completely until the next reboot or change of relevant settings.

rodhammond, 2023-07-11:
Thanks for the detailed response. I must somehow be in the second condition with aggressive NTP retry even when the device has the correct time and a network connection. For now, I manually enable after rebooting and disable once time is correct.

Aircoookie, 2023-07-28:
You can also do any main UI interaction (e.g. power on/off) to set the time from your client device, so NTP is not even required.