wled/WLED #3287: Sync specific preset or time macro from one device to another

rodhammond, 2023-07-10:
The sync feature works great - Thanks for the hard work!
Adding to the sync capability, is it possible to implement a feature where a user could select (any or multiple) presets and/or time macros to sync to specific other device? I manually do this by editing json files.

Aircoookie, 2023-07-11:
One potential issue here is different units having potentially different lenghts and thus presets might not be transferrable 1:1.
If segments would be adjusted automatically (either bounds are not included in the sync, or segments exceeding the lenght are ignored), I agree that having the capability to "share" a preset as JSON to another unit easily would be a nice enhancement.

blazoncek, 2023-07-11:
*Syncing* presets across devices is as easy as making backup and restore in Settings.
If this is useful depends how those presets are made (TL;DR).

Syncing time schedules unfortunately is not easy. If you want to give it a go, check how LED config can be transfered from device to device by uploading cfg.json to LED settings page.

rodhammond, 2023-07-11:
Yes I agree. I manually edit the json files and execute a search and replace to adjust segment lengths. Presents with multiple segments, I just divide length by number of segments. Not really that complicated to do a present sync if segment lengths are treated as equal parts of the whole. The user must understand if they do not want equal segments lengths, then adjust after the transfer. Still faster than manually editing json files. Once all presents are equal, manually editing the macro section of the config json file is pretty simple as the presents are in the same order on all of my devices.

blazoncek, 2023-07-12:
The part "user must understand" is not really something a broader public will accept.
If there is a solution it has to be fully automatic. Or else we get questions, complaints and issue reports.

So, in short, if you want the solution you'll need to write your own (and make PR if you think others may like it) or use existing approach by making a script that downloads presets from one master device and *upload*s them to other devices using existing functionality from backup and restore.

closed 2023-09-28 by blazoncek