wled/WLED #3290: wifi connectivity enhancements (multiple related)

jmerc77, 2023-07-12:
**Is your feature request related to a problem? Please describe.**
one idea is. the wifi is a bit weak.

**Describe the solution you'd like**
features:
1. force txPower to max value
2. turn off AP but turn it on when router is not connected.
3. allow backup routers to be configed.
4. use https for router connections

**Describe alternatives you've considered**
num 1 would help alot.
num 2 is mainly for security (seems there is no way to have https on softAP or maybe others)
num 4 may be silly for some leds. however, there may be some private info getting leaked. though it would also help keep people from messing with the lil' lights.
num 3 is sort of redundant. but having a plan b is great! or even c.

**Additional context**
i am working on my first esp32 project and found out some ways to do the above. assuming you are using wifi.h and wifiap.h etc.

### force txPower to max value

the wifi lib has a function to set TxPower
you will need a preprocessor branch to choose what max means per module.

### turn off AP but turn it on when router is not connected.

similarly there is a function to enable/disable that. for the second part, simply check wifi.status() and set the enable flag approriatly.

### allow backup routers to be configed.

this one baffWLED me for a few. lol. when i was messing with esp32_https_server, i discovered that wifimulti was incompatible (i think).
then i tried something stupidly silly:

```
#define AP_COUNT 2
const char* ssids[AP_COUNT]={"ssid1","ssid2"};
const char* psks[AP_COUNT ]{"psk1","psk2"};
int ap=0
//...
  //in setup...
  Serial.println("Setting up WiFi");
  WiFi.setTxPower(WIFI_POWER_19_5dBm);
  //duplicate this while stuff to loop
  while (WiFi.status() != WL_CONNECTED) {
    WiFi.begin(ssids[ap], psks[ap]);
    Serial.print(".");
    delay(500);
    ap=(ap+1)%AP_COUNT;
  }
  Serial.println();
  Serial.print("Connected. IP=");
  Serial.println(WiFi.localIP());

  while(!MDNS.begin("WLED ex")) {
    Serial.println("Starting mDNS...");
    delay(1000);
  }
```

basically "re-begaininging" the wifi with different ssids an psks.
it worked. you may also see the tx power there (only for esp32 wroom in the example).

### use https for router connections

try this: [esp32_https_server](https://github.com/fhessel/esp32_https_server)
there is a known bug. but easy to fix (it worked for me):

[esp32_https_server pr](https://github.com/fhessel/esp32_https_server/pull/164/files)
currently not merged yet.

PS: there are multiple ideas here. but is to make it easier for me to reference as they are all related both to topic and to each other.

jmerc77, 2023-07-12:
i case you did not read the above, number 2 is already in place and can be ignored.

blazoncek, 2024-07-26:
All features, except HTTPS which is infeasible, are available in 0_15 branch.

closed 2024-10-27 by blazoncek