wled/WLED #3292: Syntax Error Unexpected "<" after going to ' To the controls' section

akohli-86, 2023-07-15:
### What happened?

Get an error - syntax Error Unexpected< after opening to the control page (/sliders)

### To Reproduce Bug

Go the index page and click on to the controls. Setting work fine but 'To the Controls' is messed

### Expected Behavior

Lower lage will show error

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-07-15:
See 
* https://github.com/Aircoookie/WLED/issues/3259

It looks like the problem is already fixed in latest source code.

akohli-86, 2023-07-16:
Tried the fix,  didn't help.  That's why i have raised it again...

softhack007, 2023-07-16:
Did you perform `npm run build` after changing  index.js ?

https://kno.wled.ge/advanced/custom-features/#changing-web-ui

blazoncek, 2023-07-16:
The original problem is solved in the source. Please use most recent nightly build (not a release) or compile from source yourself.

closed 2023-07-17 by akohli-86

koenieee, 2023-08-06:
Can someone please start a new release without the bug? NIghtlies aren't avaible for the ESP01 on https://wled-install.github.io/