wled/WLED #3294: Relay GPIO setting resets after upload output from xLights

Webguyatwork, 2023-07-17:
### What happened?

I recently added a relay to my Dig Quad controller panel using GPIO pin 15 as the trigger. WLED was also set up with pin 15 under LED Preferences. If I do nothing more other than test with WLED itself, this setup works great.

Switching to xLights, after mapping models to the appropriate ports on the Dig Quad and pushing the outputs to the controller, the Relay GPIO pin gets automatically set back to "Unused", rendering the controller inoperable as far as pushing pixel data goes.

### To Reproduce Bug

**Reproducing this bug does not require the use of a relay to switch the power supply, but it does help to ensure the configuration is working properly.**

**Start in WLED:**
- Go to the LED Preferences configuration page. For the Relay GPIO setting, select pin 15. (Noted, this pin is dual purpose on the Dig Quad, allowing for pushing pixel data OR signaling a relay)
- Save the configuration and test with the power button, making sure lights turn on and off appropriately
- Check under LED Preferences again to ensure that the Relay GPIO pin has not been changed.

**Now switch to xLights:**
- Open any valid layout with the controller defined as
  - Vendor: WLED
  - Model: QuinLED-Dig-Quad
  - Variant: ESP32
- Map any number of LEDs to one of the controller outputs in the visualizer
- Click on upload outputs
- Back in WLED, go back into the LED Preferences tab and notice that the Relay GPIO setting is back to "Unused"

### Expected Behavior

Assuming the Relay GPIO has been set within WLED, it is expected that this setting will not be initialized by the uploading of outputs from xLights

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b3 (build 2306210)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-07-17:
This is not a WLED bug, xLights (and other software) can configure WLED.
Please research your light control software how to properly configure WLED.

Webguyatwork, 2023-07-17:
Thanks for that response. I debated whether this was an issue of the data being sent from xLights, or the way WLED was interpreting and applying it. Sounds like a network trace is in order.

Webguyatwork, 2023-07-17:
I have more information now from a packet capture. I see a GET call to /json/cfg where pin 15 is set correctly:

```
"relay": {
      "pin": 15,
      "rev": false
 },
```

And then a subsequent POST to /cfg/json where this same relay configuration is still intact. So it's not xLights sending a reset after all. I'm back to believing it's in the way WLED is handling the POST data.

blazoncek, 2023-07-18:
This will not work as configuration only has a logic to allocate GPIO not re-allocate it (even though it may be the same).
Relevant code:
```c++
  JsonObject relay = hw[F("relay")];
  int hw_relay_pin = relay["pin"] | -2;
  if (hw_relay_pin > -2) {
    if (pinManager.allocatePin(hw_relay_pin,true, PinOwner::Relay)) {
      rlyPin = hw_relay_pin;
      pinMode(rlyPin, OUTPUT);
    } else {
      rlyPin = -1;
    }
  }
  if (relay.containsKey("rev")) {
    rlyMde = !relay["rev"];
  }
```
So you need to exclude relay GPIO from xLights.

I do not consider this to be a bug in WLED but rather a design choice. Still, as the configuration API was written by @Aircoookie let's hear his opinion.

Aircoookie, 2023-07-19:
I would definitely consider this behavior a bug. `pinManager` rejecting re-allocation is indeed a design choice, to e.g. prevent `PinOwner::Button` from allocating the same pin twice for multiple buttons. However, `cfg.cpp` should deallocate the previous button pin before attempting to allocate again. Will add that, similarly this fix is also required for IR and button pins.

blazoncek, 2023-07-19:
IMO this will add unnecessary complexity to `cfg.cpp`. IMO `cfg.cpp`'s intention is clear: Load configuration only or save it only.
It is `set.cpp`'s duty to reconfigure WLED device (and cause saving of configuration using `cfg.cpp`). It may be better to redirect re-configuration to `set.cpp` source.

Aircoookie, 2023-07-19:
I wouldn't say so, deserializing JSON from `/json/cfg` is a supported use case right now and the new settings page (if I get around to it in this life ^^) will use `cfg.cpp`, _replacing_ `set.cpp`.

Webguyatwork, 2023-07-20:
This does indeed appear to resolve the issue. I can appreciate what @blazoncek is saying about how the light software is sending the updates through, but it's just doing a GET config, update only what it needs to, and the POST the updated config which still includes the relay config along with everything else.

Thanks for the super quick turnaround on this fix!

blazoncek, 2023-08-05:
@Webguyatwork please close if your issue is solved.

Webguyatwork, 2023-08-12:
I just updated my primary controller to WLED 0.14.0-b4 (ESP32-D0WDQ5 build 2308110) as released today. This fix works perfectly. Thanks again!!!

closed 2023-08-12 by Webguyatwork